MarlinFirmware/Marlin #7326: [FR] Improved Z-Offset babystepping screen

marcio-ao, 2017-07-20:
The LulzBot branch of Marlin has a graphical overlay for babystepping the Z-Offset. This helps the user understand which direction to turn the dial:

[image]

I've recently been porting our changes to Marlin 1.1.4 and would like to ask whether this is something that would be welcomed upstream. The downside is that it does require extra memory for the graphics and currently it will not look good if USE_BIG_EDIT_FONT is enabled. Making it work for both cases would probably require a customized version of "lcd_implementation_drawedit" just for the Z-Offset, so it would require even more memory.

So I am posting this here to learn whether there is an interest in incorporating this. If so, I could make a push request.

gcormier, 2017-07-21:
I think it would be a great option to have, and I'd use it!

Roxy-3D, 2017-07-21:
> I think it would be a great option to have, and I'd use it!

Yes...   BUT!!!!!   It needs to draw the clockwise and counter-clockwise pictures accurately based on the values of
```cpp
//#define REVERSE_ENCODER_DIRECTION
//#define REVERSE_MENU_DIRECTION
```
We can't have it saying to do the wrong thing!!!!

marcio-ao, 2017-07-21:
Good point. Should not be hard to do. I'll see if I can put together a pull request.

Roxy-3D, 2017-07-21:
> Good point. Should not be hard to do. I'll see if I can put together a pull request.

I don't know what the 'right' answer is.   But maybe we could use a 'normalization' function within the LCD Panel.   One of the things on the list would be Encoder Wheel direction.   And in this sub-function, it will confirm the firmware is seeing the right thing when the wheel is turned one way or the other.

fiveangle, 2017-07-24:
My PB with MINIPANEL uses REVERSE_ENCODER_DIRECTION (and not REVERSE_MENU_DIRECTION) in case you get to the point of needing to verify PR working correctly on acual HW (athough I think one could just test with standard HW and ensure the value corresponds to opposite when set so).

fiveangle, 2017-07-24:
While you're at it, change BABYSTEPPING to LIVE_Z_ADJUST ;)

Roxy-3D, 2017-07-24:
> While you're at it, change BABYSTEPPING to LIVE_Z_ADJUST ;)

And leave the X & Y axis BabyStepping as BabyStepping?  Or do those turn into LIVE_X_ADJUST and LIVE_Y_ADJUST ?

fiveangle, 2017-07-24:
Even LIVE_AXIS_ADJUST would be clearer (even though it's typically used for mostly Z).

Prusa MK2 fork doesn't implement X or Y live adjustment (didn't know Marlin does, but not surprised) so they call it LIVE_Z_ADJUST and as expected, no one seems confused by it.  However, even for text displays, "Clockwise", "Counter-clockwise" would be nice additions.

fiveangle, 2017-07-24:
Just checked and yeah while a print I just started and yup,, Prusa doesn't have direction marked either. You know which way you are supposed to go, but when the filament is curing up into the HE and you're scrambling to grab the live z, you definitely have to stop and think a second which direction to move just because you know you only have a few seconds before having to scrap the whole print and start over :)

http://www.youtube.com/watch?v=DtrSQ4yW1m0

marcio-ao, 2017-07-24:
I have created a pull request for this: #7350. At the present, it will only show up in the Z-offset screen, but if desired, it could trivially be added to all the edit screens to indicate to the user which direction to turn the knob to raise or lower values (if doing this, I would rename _lcd_babystep_zoffset_overlay to something else and add an argument to show/hide the nozzle picture, which only makes sense for z-offsets).

marcio-ao, 2017-07-24:
Probably a better way to implement this would be to put the overlay in "lcd_implementation_drawedit" and to shift the text up to make room for the arrows. This would make it so the arrows need not be scrunched together on the left when USE_BIG_EDIT_FONT was enabled.

But I don't quite get how the centering code in "lcd_implementation_drawedit" works, nor do I really know whether it would be desirable to have the indicators show up on all edit controls, which is why I implemented it as an overlay in my patch.

petrzjunior, 2017-07-25:
Speaking of babystepping, Průša has an interesting feature: saving Z offset. You cas edit it from LCD just like babystepping and save it into EEPROM. This way you manually compensate the bed-nozzle gap at any time.

gcormier, 2017-07-25:
@petrzjunior That's available in Marlin as well.

fiveangle, 2017-07-25:
@gcormier - are you talking about BABYSTEPPING ?

FHeilmann, 2017-07-25:
@fiveangle hes talking about the BABYSTEP_Z_OFFSET option

fiveangle, 2017-07-25:
so confusing

Redferne, 2017-07-26:
@petrzjunior

Where can I find it? I always like to finetune the first layer by double-click and adjust Z. But starting a new print it always reverts to 0.

gcormier, 2017-08-29:
@marcio-ao Did this PR ever happen?

marcio-ao, 2017-08-29:
@gcormier: As far as I know, no.

gcormier, 2017-08-29:
Okay. Still interest from me :)

marcio-ao, 2017-08-30:
I've resubmitted pull request with feedback

https://github.com/MarlinFirmware/Marlin/pull/7586

marcio-ao, 2017-09-05:
Closing as this has been merged.

closed 2017-09-05 by marcio-ao