MarlinFirmware/Marlin #7327: [FR] Hold button to show About page.

marcio-ao, 2017-07-20:
Currently if you push the LCD button while on the info screen, the splash screen flashes momentarily. We often use this to check the firmware version, but this is an unintended feature (i.e. bug). It flashes really quickly, making it hard to read the version number.

How about making this bug into a feature so that holding down the LCD button keeps the About screen up until you release it? :)

Tannoo, 2017-07-22:
What LCD do you have and could you post your configuration file?

I can't reproduce what you are seeing.
Firmware version is displayed by MENU > ABOUT PRINTER > PRINTER INFO.

Tannoo, 2017-07-22:
If I hold mine from the status screen, it just cycles from status, main, z-offset, status, main, z-offset, ....

Tannoo, 2017-07-22:
If Marlin is resetting while trying to go to the menu, there is a voltage drop on the 5v rail. Enough to cause the CPU to restart.

marcio-ao, 2017-07-24:
Hum. I don't have an "About Printer" under Marlin 1.1.4 stable. It does flash the startup screen momentarily when I press the button, right before showing the menu. It's very fast though, you can barely see it.

Tannoo, 2017-07-24:
Try rcbugfix-1.1.x.

Tannoo, 2017-07-26:
> I don't have an "About Printer" under Marlin 1.1.4 stable.

Oh... I have `LCD_INFO_MENU` enabled.
I will disable that and try again.

Tannoo, 2017-07-26:
Yeah... I still can't reproduce your issue with bugfix-1.1.x

marcio-ao, 2017-07-26:
Ah. It only happens if SHOW_CUSTOM_BOOTSCREEN is enabled.

Tannoo, 2017-07-26:
Do you have a custom bootscreen?

marcio-ao, 2017-07-26:
Let me explain exactly what I am trying to do. I modified the lcd_custom_bootscreen() so it shows some additional information about our printers. I kept the Marlin splash screen the same. So currently at startup I get this screen:

[image]

Followed by this screen:

[image]

The bug is that the second screen flashes momentarily when I push the encoder wheel. Not sure why this happens. It seems to be a bug.

But what I would *like* to have is actually have a menu option that shows the first splash screen until the user clicks the button again. I could modify our firmware to do this if someone would give me a few pointers on how to do it. I can't imagine it would be very hard to do.

Thanks.

marcio-ao, 2017-07-26:
@Tannoo: Yes. I do. I think this issue request may have been confusing because I was mistakenly referring to the lcd_custom_bootscreen as an About screen. I wasn't aware (until today) of LCD_INFO_MENU, which I guess is what you guys are calling the "About printer" screen.

I just tried LCD_INFO_MENU and I understand better the distinction now. I would still like to be able to call up our customized lcd_custom_bootscreen again.

fiveangle, 2017-07-27:
Sounds like this could be easily implemented by overloading the LCD_INFO_MENU to cycle between what it shows now, and the custom boot screen periodically.

Tannoo, 2017-07-27:
How about another info option to show the custom boot screen?

marcio-ao, 2017-07-27:
@Tannoo: That would be great, it's what I am looking for.

@fiveangle: The code you quoted appears to be function prototypes. I would be surprised if that bit of code was generating the menus. Is there any document anywhere giving a high level overview on how menus are defined?

marcio-ao, 2017-07-27:
I was able to implement the custom menu entry to do what I wanted by following the examples in "ultralcd.cpp". Now I have an "About Printer" menu, but it shows the custom splash screen rather than showing a submenu. I will close this ticket since I have accomplished what I needed.

closed 2017-07-27 by marcio-ao