MarlinFirmware/Marlin #7330: Odd Printer Behavior

devekrehbiel, 2017-07-21:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

devekrehbiel, 2017-07-21:
I upgraded to the latest Marlin about 3-4 weeks ago and also upgraded to Cura 2.6 when the problem in these pictures arrived. It could very well be my inexperience but I am desperately hoping someone here can help. I went to an older Cura but it didnt help. I am running an Anet A8.  
Here are 8 pictures of the problem. What am I doing wrong?
http://devestechnet.com/Images/TopLayerIssue01.jpg
http://devestechnet.com/Images/TopLayerIssue02.jpg
http://devestechnet.com/Images/TopLayerIssue03.jpg
http://devestechnet.com/Images/TopLayerIssue04.jpg
http://devestechnet.com/Images/TopLayerIssue05.jpg
http://devestechnet.com/Images/TopLayerIssue06.jpg
http://devestechnet.com/Images/TopLayerIssue07.jpg
http://devestechnet.com/Images/TopLayerIssue08.jpg
Is it the slicer or the sketch or me?

thinkyhead, 2018-02-09:
Keep adjusting those slicer settings.

closed 2018-02-09 by thinkyhead