MarlinFirmware/Marlin #7331: UBL mesh seems "tilted" in automatically-probed area?

hoppersoft, 2017-07-21:
Monoprice Maker Select v2.1
LJ18A3-8-Z probe (Z offset -1.6mm, X 40, Y 48)
Marlin 1.1.3 (b2c20a8c)

Hello! I have installed an inductive probe and have configured Marlin to use it. G28 followed by a G0 Z0 results in a nice drag on a piece of paper at X100 Y100 Z0 post-homing. However, when I follow the UBL process described [here](http://marlinfw.org/docs/features/unified_bed_leveling) I get a mesh that seems perfect wherever I manually probed but is "tilted" for the automatic probe. After following the process and activating UBL, moving the probe to the front-left (towards X0 Y0) of the auto-probed area results in the nozzle hitting the bed and towards the rear-right (X200 Y200) of the probed area the nozzle doesn't grip the paper at all.

If you were to build a surface chart of the below, you may see what I'm referring to; there's a jump of nearly a layer's worth between the boundary, and it's evident when you run a G26. Can anyone offer any possible reasons for why it could be behaving this way?

Many thanks for reading this far!

Here's the matrix I have. The automatically probed zone is within the set of coordinates:
3,7 - 3,9
3,0 - 3,9

(Don't pay too much attention to the 0,9 and 9,0 corners)
`
(0,9)                                                                             (9,9)
(1,199)                                                                         (199,199)
 0.040   -0.080   -0.100   -0.080   -0.050   -0.080   -0.100   -0.140   -0.150   -0.190

-0.140   -0.080   -0.080   -0.030   -0.040   -0.030   -0.070   -0.100   -0.110   -0.150

-0.140   -0.020   -0.040   -0.095   -0.038    0.020    0.060    0.047    0.017   -0.010

-0.080    0.000   -0.020   -0.105   -0.035    0.017    0.055    0.055    0.025   -0.010

-0.030    0.020   -0.030   -0.103  [-0.040]   0.017    0.040    0.047    0.022   -0.015

-0.020    0.000    0.000   -0.105   -0.045    0.002    0.030    0.020    0.002   -0.040

-0.070    0.000   -0.010   -0.115   -0.048   -0.003    0.000    0.007   -0.033   -0.070

-0.060   -0.060   -0.040   -0.145   -0.105   -0.055   -0.038   -0.035   -0.063   -0.105

-0.120   -0.040   -0.080   -0.200   -0.145   -0.090   -0.075   -0.070   -0.083   -0.113

-0.140   -0.080   -0.080   -0.225   -0.177   -0.115   -0.105   -0.088   -0.070    0.055
(0,0)                                                                             (9,0)
(1,1)                                                                            (199,1)
`

Roxy-3D, 2017-07-21:
I'll re-read this in the morning.  And maybe plot the points that G29 T1 gives me in Excel.

But is it possible your bed really is tilted?   That would be my guess.

hoppersoft, 2017-07-21:
I'm sure it's possible (although I did spend quite a bit of time manually
leveling it using the paper technique at all four corners). That being
said, if the bed itself were tilted, shouldn't UBL use the offsets to
account for that? What I'm seeing during the print is that it doesn't
appear to be (and I did run G29 A and M500 to save it prior to running the
G26 test print). In addition, the big difference at the boundary between
the manually-probed and automatically-probed areas just doesn't look right.
It's a big jump for what appears to be a pretty flat bed.

Roxy-3D, 2017-07-21:
> It's a big jump for what appears to be a pretty flat bed

I don't know...    The whole reason Patel developed the Mesh Bed Leveling system was no matter what he did...   No bed was really flat.   When you look across a piece of really 'flat glass', you still see imperfections.

The whole trick is to get a first layer of adhesions you can build upon.   And if you can get that...   You are going to get a print.

BillyBob54, 2017-07-21:
Is this the same thing that issue discussed in #7278 ?

Roxy-3D, 2017-07-21:
It is possible....  But it would be good to see some of the Topography maps (both before and after a G29 P3).

BillyBob54, 2017-07-21:
Currently on the road, I'll see if I can get to it this weekend.  I'll zero everything out and repeat the leveling process.  I only use a 5X5 matrix and it's pretty quick to dial everything in.

hoppersoft, 2017-07-22:
Buying my wife a car right now, I'll do the same as soon as I get home!

hoppersoft, 2017-07-22:
Here's a G29 T post-P1 with the inductive probe:

`Bed Topography Report:
(0,9)                                                                   (9,9)
(1,199)                                                                (199,199)
     .          .          .          .          .          .          .          .          .          .

     .          .          .          .          .          .          .          .          .          .

     .          .          .      -0.067   -0.003    0.052    0.072    0.063    0.032   -0.008

     .          .          .      -0.075   -0.025    0.040    0.063    0.060    0.037   -0.005

     .          .          .      -0.070  [-0.030]   0.037    0.060    0.050    0.032   -0.010

     .          .          .      -0.095   -0.050    0.025    0.037    0.040    0.025   -0.020

     .          .          .      -0.125   -0.063   -0.005    0.027    0.030    0.000   -0.038

     .          .          .      -0.168   -0.098   -0.043   -0.010   -0.008   -0.013   -0.045

     .          .          .      -0.210   -0.130   -0.067   -0.045   -0.028   -0.020   -0.023

     .          .          .      -0.890   -0.823   -0.763   -0.790   -0.780   -0.795   -0.635
(1,1)                                                                    (199,1)
(0,0)                                                                     (9,0)
`

And here's the same map post P3:
`G29 P2 finished.

Bed Topography Report:
(0,9)                                                                   (9,9)
(1,199)                                                                (199,199)
  0.040    0.080    0.060    0.110    0.100    0.080    0.020    0.000    0.010  [-0.100]

  0.100    0.080    0.060    0.120    0.120    0.180    0.040    0.060   -0.040   -0.070

  0.100    0.140    0.110   -0.067   -0.003    0.052    0.072    0.063    0.032   -0.008

  0.180    0.160    0.180   -0.075   -0.025    0.040    0.063    0.060    0.037   -0.005

  0.180    0.220    0.160   -0.070   -0.030    0.037    0.060    0.050    0.032   -0.010

  0.200    0.200    0.200   -0.095   -0.050    0.025    0.037    0.040    0.025   -0.020

  0.160    0.200    0.180   -0.125   -0.063   -0.005    0.027    0.030    0.000   -0.038

  0.100    0.100    0.080   -0.168   -0.098   -0.043   -0.010   -0.008   -0.013   -0.045

  0.080    0.100    0.040   -0.210   -0.130   -0.067   -0.045   -0.028   -0.020   -0.023

  0.130    0.060    0.020   -0.890   -0.823   -0.763   -0.790   -0.780   -0.795   -0.635
(1,1)                                                                    (199,1)
(0,0)                                                                     (9,0)
`
This confuses me. I'm seeing a big jump as we transition from the auto-probed area to the manually-probed area. I believe I have the z-offset set correctly, as disabling UBL, followed by a G28 followed by moving to Z0 results in a nozzle that barely drags the paper. I'm following the steps outlined in http://marlinfw.org/docs/features/unified_bed_leveling.html. Am I missing something?

hoppersoft, 2017-07-23:
Hmm. From what I've been reading it sounds like having my X/Y probe offsets off by too much might result in this. I've been using the distance in the X/Y axis from the middle of the probe to the tip of the extruder's nozzle; is that what I should be using?

Roxy-3D, 2017-07-23:
> Hmm. From what I've been reading it sounds like having my X/Y probe offsets off by too much might result in this. I've been using the distance in the X/Y axis from the middle of the probe to the tip of the extruder's nozzle; is that what I should be using?

Yes.  But ideally, you want to probe very close to the nozzle.

The 'big jump' is not really that big.  Your bed is very flat.  You should be able to do a G26 to see how good the filled in (not probed) locations are.  And it is very quick to edit them if they need to be adjusted.

hoppersoft, 2017-07-23:
Yeah, the locations where I manually adjusted during P3 are great; it's
just the locations that were auto-probed that seem to have this issue. I'm
going to triple-check my distances to see if there's any chance I'm off.

apballard, 2017-07-23:
Hi, I'm not sure if I'm experiencing the same issue or if it different.

Normally I print smaller items in the middle of my build plate, so until recently i have not noticed this problem.  Now that I'm printing a large object nearly the full width of my build plate, I've noticed a problem.  Not sure if it mechanical or firmware.

I'm using an IR sensor, and the points automatically scanned on the sides/perimeter on the x axis (the left and right) are always out.  G26 results in nothing printing in those areas because the nozzle is too close to the bed.  However all the points printed in the middle and at the back and front are fine.

I just thought being so close to the edge of the build plate, it may be that my IR sensor is aggregating the height in an area that includes an area off the build plate, but that is not the case.  I edited the mesh area to bring in the sides by 1cm and am still having the same issue.

Is this a similar problem?

cisforcer, 2017-07-23:
[image] same to me……run g26, show  out of  area , what's  my
 missing?

Roxy-3D, 2017-07-23:
Well...   That is the purpose of the G26 command.   It lets you see how well the mesh points reflect reality.   Using that information, it is pretty easy to edit the mesh points and get them to be perfect.

If you do a G26 and post a picture of the Mesh Validation Pattern...   It will be easy to see what needs to be done.

Roxy-3D, 2017-07-23:
> I'm using an IR sensor, and the points automatically scanned on the sides/perimeter on the x axis (the left and right) are always out. G26 results in nothing printing in those areas because the [...]

So...   edit those areas...   And push those mesh points down .1mm.     And then do another G26.   If things aren't perfect...  Move the mesh points some more...      Wash... Rinse...  Repeat...    It should only take 2 or 3 iterations and you will have use of the entire bed area.

Roxy-3D, 2017-07-23:
> same to me……run g26, show out of area , what's my missing?

That doesn't look like a G26 Mesh Validation Pattern.

apballard, 2017-07-24:
> So... edit those

@Roxy-3D yes I'm, I love the fact that I can. Just wondering if it is a similar problem and if anyone else is experiencing anything similar. Just the edges that don't scan accurately.

Roxy-3D, 2017-07-24:
> yes I'm, I love the fact that I can. Just wondering if it is a similar problem and if anyone else is experiencing anything similar. Just the edges that don't scan accurately.

I'm still trying to figure things out.   The Cartesian version of G29 P3 tries to extrapolate the bed tilt to the unprobed mesh point locations.  But only if the mesh is tilted upwards.   That is a safety (for the printer) constraint.   It might result in the filled in mesh points being further from reality than they need to be.  But in that case it is done just to be safe.

The Delta version is doing a LSF on the near by mesh points.   There is a lot thinking going on to find the a simple but sufficient algorithm.   The good news is...  If you can get any thing to print...  It is very easy to edit the points and zero in on the correct numbers.    Probable it is best to just view the initial probing as the the starting place for the final mesh.   The fact some points can't be probed means there is some guess work involved with filling in the mesh.   But as long as a G26 Mesh Validation Pattern can be printed, it only takes a few iterations to get the entire bed to have complete adhesion.

closed 2018-02-09 by thinkyhead

ramireseverton, 2018-06-23:
I face the exactly same issue. Any way to tilt this wrong mesh back to a right position manually since auto-probing always returns a tilted mesh?
I get Roxy's point that using m421 would fix, but why edit like 20 points one by one when I could just force a tilt in the whole mesh since it seems like that?

BillyBob54, 2018-06-23:
I never really received an explanation for what was happenIng. The thread
sort of wandered off topic. I just fixed the mesh with m421 and went from
there. I only use a 5x5 mesh so it goes pretty quickly. Still, I always
thought there must be a fault in the algorithm as by all rights it should
be pretty much perfect out of starting gate. As far as I know, there is no
way to just tilt the mesh

thinkyhead, 2018-06-26:
If the mesh is tilted compared to the bed, this usually indicates that the probe is not staying perfectly straight as it moves in the apparently-tilted direction. A twist in the X-gantry will cause this, and we see this issue at epidemic levels with Chinese clones built using 4020 and 2020 extrusions.

With UBL you can use `G29 J` (See http://marlinfw.org/docs/gcode/G029-ubl.html) to tilt the mesh based on three probe points. But this will actually tilt the mesh more, rather than un-tilt it based on the assumption of bad probe alignment.

All bed leveling methods assume the probe is physically adjusted so it remains always at the same angle to the bed. Unfortunately we have no way to tell when a probe is twisting as it moves and giving bunk values.

BillyBob54, 2018-06-27:
Makes sense. All the wiring runs to the right side of the printer and I
always thought that it might have been pulling on the print head. Visually
I could never see anything and the wires never seemed to pull. Twisted X
gantry might be, just odd that it seems to always manifest on the left
side. One of the great mysteries of the universe.

lrpirlet, 2018-06-28:
@BillyBob54 

> Twisted X gantry might be, just odd that it seems to always manifest on the left side. One of the great mysteries of the universe.