MarlinFirmware/Marlin #7333: UBL -- Nozzle crashes into bed in between prints when G28 issued in start script.

aviatorken89, 2017-07-21:
I've been having a heck of a time trying to get UBL working reliably the past couple of days, I had several crashes trying to understand the documentation and offsets, but I managed to generate a 15x15 mesh automatically.

Before I used bilinear leveling before every print, and my gcode start script before each print looked like this:

> M190 S[bed0_temperature] ; Set bed temperature and wait
> G28 ; Home all axes
> G29 ; Auto bed level
> M117 Printing ; Display 'Printing' message

After enabling and activating UBL and saving to EEPROM and from what I gathered from the confusing documentation I changed my start script to this:

> M190 S[bed0_temperature] ; Set bed temperature and wait
> G28 ; Home all axes
> M117 Printing ; Display 'Printing' message

The first print on powerup goes fine, but if I start another print without power cycling then the G28 proceeds to crash into the bed.

Can someone help me figure this out? It may be a good idea to add simple script documentation to the UBL pages because it isn't user friendly by any means. Even with several googlings I came up with not much or any info on the subject.

Thank you!

Roxy-3D, 2017-07-21:
What is the purpose of the G29 with no parameters on it?    I don't think that is causing the bad behavior.   But that serves no purpose.

aviatorken89, 2017-07-21:
If you read my post carefully you will see the G29 without parameters was from when I had Bilinear enabled before upgrading to UBL. Do I need M420 S1 before G28 now?

Thanks!

aviatorken89, 2017-07-21:
Update: I can confirm it isn't a Marlin issue, it's the BLTouch being way too high, despite reading their documentation.

Sorry for opening this issue :)

closed 2017-07-21 by aviatorken89