MarlinFirmware/Marlin #7335: Marlin Delta Homing Issue

yashunandansureka, 2017-07-21:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

### Bug Report
- Description: --- The Delta axis hits the endstop and crashes into it. Shows crazy behaviors. 
- Expected behaviour: --- The axis should home on G28
- Actual behaviour: --- The axix go up and crash into the endshop. The Belts start to vibrate crazy.
- Steps to reproduce: G28, Should home correctly
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1164661/config.zip)

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

teemuatlut, 2017-07-21:
Verify your endstops are working by using M119.

yashunandansureka, 2017-07-21:
I tried that. It is all weird. 

I have two deltas. One of them runs older version of marlin and actually works pretty well.

This second delta i recently made, i wanted to put the latest marlin with auto bed leveling.
Somehow it just does not work.

I have noticed that in the new Marlin fw, 1.1.4, there are a lot a settings regarding delta printers that are missing,such as DELTA_CARRIGE_OFFSET, DETLA_EFFECTOR_OFFSET. This also left me a bit puzzled.

RATT88, 2017-07-24:
Yes. This is what is happening. I can rectify it temporarily in repetier host by hitting all home and then setting home. I thought it was sorted but when I came back to it the next day (and didn't go through repetier) and tried to just print from sd card the problem returned. It's as if it's defaulting to something buried in the firmware code.

RATT88, 2017-07-24:
Also ran m119 and verified that all switches are working.

yashunandansureka, 2017-07-24:
Yeah, some times it works, some times it doesn't.
I'm thinking of changing the wiring of the end-stops and the end-stops too.

Will keep ya'll posted.

yashunandansureka, 2017-07-26:
Okay, Update.

The situation seems to be solved. I changed the endstop wiring. 

The z and x axis edstops were problematic but now they seem to work fine. The x axis is a little but moody sometimes but asof now it seems ok.

RATT88, 2017-07-26:
Cool. I might try this. How exactly did you change the wiring?

yashunandansureka, 2017-07-27:
I changed the wires that were going from the endstops to the Ramps.

ScanmanIR, 2017-08-01:
I believe that I have a similar problem on a Hatchbox Alpha Delta printer. My Z Axis hits the stop at the home position and continues to run with the belt shaking until we turn it off.  I am not too familiar with coding, but I come from an electronics background.  I have family that can code.  What kind of diagnostics can I run to check out the system?

yashunandansureka, 2017-08-09:
Hi,

Try changing the wires that connect the endstop. I hope that works.

BTW, sending M119 to the printer will give you an echo of which endstop is triggered and which is not, you can use it to see if the endstops work properly or not.

closed 2018-02-09 by thinkyhead