MarlinFirmware/Marlin #7336: [Question] Emulate trigger endstop with code for custom IR sensor

raguera, 2017-07-21:
Hi all,

As I have a glass over the aluminum heated bed in my printer, capacitive or inductive sensors are not useful for me, so I want to implement an IR sensor to level the heated bed.

I am writing a piece of code on temperature.cpp, as it needs to be called constantly to read the IR value through an analog pin (A9 in my case). It shows the value propperly, but the problem now is, when it reaches certain value, what function/variable I have to set to "TRUE" or "ENABLE" to emulate an endstop trigger?

I didn't find any related and maybe it could be a pull request if it's not yet implemented.

Thanks a lot

Bob-the-Kuhn, 2017-07-24:
You need to be working within the routine **Endstops::update()**.  That's the routine that checks the endstops, stops the steppers when the appropriate endstop is hit and updates the flags that tells the world that an endstop has been hit.

thinkyhead, 2018-02-09:
I'm working with a client who may include IR sensor support in their product. When they ship the code will be public. But if you already have something working then you may submit a PR any time.

boelle, 2019-02-19:
@raguera do you have code that can be submitted as PR?