MarlinFirmware/Marlin #7342: [FR] Distracting Flashing Question Marks

thomllama, 2017-07-22:
OK, I know why the things are there and flashing between the XYZ and the question mark. But I find it extremely distracting and 100% of the time I home everything before printing or doing anything at all so having the warning isn't really nessesery. I fully understand that it's a good thing for many people, but for me I much prefer it not flash, it's distracting when I keep seeing it out of the corner of my eye while trying to do other things and the printer is idle 

I heard turning _"DISABLE_REDUCED_ACCURACY_WARNING"_ on would stop it, I tried,  it didn't. 

Can we just please stop the flashing.. make the actual read out a question mark instead of being zero and leave the XYZ alone.. and no flashing?   :)  It would be most awesome that way. 

If there is a way to stop it I've missed let me know please.   I know you guys work pretty hard on all this and it's MUCH appreciated even if you don't always hear it. !!! Thanks

thinkyhead, 2018-02-09:
The `DISABLE_REDUCED_ACCURACY_WARNING` only turns off the blinking behavior that occurs when steppers go to sleep. The blinking will still occur before you issue a `G28`.

To stop the blinking question marks, do `G28` as soon as you power up.

To eliminate them, modify this:
```cpp
FORCE_INLINE void _draw_axis_label(const AxisEnum axis, const char* const pstr, const bool blink) {
  if (blink)
    lcd_printPGM(pstr);
  else {
    if (!axis_homed[axis])
      u8g.print('?');
    else {
      #if DISABLED(HOME_AFTER_DEACTIVATE) && DISABLED(DISABLE_REDUCED_ACCURACY_WARNING)
        if (!axis_known_position[axis])
          u8g.print(' ');
        else
      #endif
          lcd_printPGM(pstr);
    }
  }
}
```

closed 2018-02-09 by thinkyhead

KrX3D, 2019-09-19:
@thinkyhead 

that did only partly work for me. i could see the number but not "X Y Z" anymore. 
so i changed for Marlin 2.0 at \Marlin\src\lcd\dogm\status_screen_DOGM.cpp

the line you mentioned to:

`FORCE_INLINE void _draw_axis_value(const AxisEnum axis, const char *value, const bool blink) {
  const uint8_t offs = (XYZ_SPACING) * axis;
  lcd_put_wchar(X_LABEL_POS + offs, XYZ_BASELINE, 'X' + axis);
  lcd_moveto(X_VALUE_POS + offs, XYZ_BASELINE);
  if (blink)
    lcd_put_u8str(value);
  else {
      #if NONE(HOME_AFTER_DEACTIVATE, DISABLE_REDUCED_ACCURACY_WARNING)
        if (!TEST(axis_known_position, axis))
          lcd_put_u8str_P(axis == Z_AXIS ? PSTR("       ") : PSTR("    "));
        else
      #endif
          lcd_put_u8str(value);    
  }
}`

it would be nice if there was an option at the configuration file to enable/disable it