MarlinFirmware/Marlin #7343: Marlin v1.1.4 advanced pause not work!

vladi552, 2017-07-22:
### Bug Report
- Description: ---
Advanced pause future (AP) not work from my G-code but working good touch the button.
- Expected behaviour: ---
Advanced pause and continue...
- Actual behaviour: ---
When pause occurs nozzle go away as expected but immediatelly returned to the print position and paused here. User execute AP touch the button. The nozzle correctly go to pause coordinates and stop here. But printing cannot be continue becourse a nozzle temp on the 'PRINT PAUSED' screen slow down and cannot heat up again. The only exit is stop the print.
- Steps to reproduce:
Look to the my G-code pcs where M125 command executed. Try to reproduce with config files included.

[Config.zip](https://github.com/MarlinFirmware/Marlin/files/1167487/Config.zip)

Roxy-3D, 2017-07-22:
Can you check if the filament change (M600) works from Gcode?   That would be helpful to know.

vladi552, 2017-07-23:
I had some test only for 'Print from SD' mode. I have the TEVO BlackWidow printer (graph LCD and encoder as a user interface). I use Simplify3D slicer.
1. The M600 cmd working well for me. But I need only smart pause to insert a nuts to the model.
2. To do so I use M125 and M0 cmd and it's also works! Printer goes to the paused state at the parking position. But in the LCD 'Tune' submenu the 'Change filament' cmd still available. If user execute this in the paused printer the only exit is a printer reset. I think, 'Change filament' must not be available at the paused printer state.
3. Pause from SD give the users only option to extrude some filament. Also it is important to do a retract before print resuming. Currently it is not possible at the paused state?

vladi552, 2017-07-23:
I need only a smart pause to insert a nuts to the model ... New test results:
1. M125 and M0 can not be used. Once M125 executed and nozzle parking to the given location the PAUSE_PARK_NOZZLE_TIMEOUT started.
2. User insert nuts. Some time needed.
2. If PAUSE_PARK_NOZZLE_TIMEOUT timeout occurs the nozzle temp is slow down.
3. To continue printing user click the button.
4. Nozzle immediatelly moved to print location and start heating here and print continues. This is wrong behavior. Nozzle must heat up at the park position.
How can I make this in Simplify3D?

closed 2017-07-26 by vladi552