MarlinFirmware/Marlin #7344: [FR] Add an example for RAMPS board

bryonbaker, 2017-07-23:
There is no example for RAMPS and a RepRapDiscount Smart Controller.
I have one on a branch ready to push if this feature is accepted.

fiveangle, 2017-07-24:
Isn't this simply defining RAMPS board, then uncommenting RepRapDiscount Smart Controller define ?  If a user is compiling Marlin (instead of installing pre-compiled from somewhere else) they would be expected to understand how to enable/disable features in Configuration.h so I'm not sure what the value add here is ?

Bergerac56, 2017-07-24:
I agree. Except a RAMPS board and the LCD, there are a lot of parameters dependent from the stepper motors, the endstops swiches and the Z probe, to take only some exemples.

If you want just try and learn, you can first, as fiveangle suggests, enable RAMPS board and RepRapDiscount Smart Controller and use the per default parameters. BUT you will have to pe careffull. The way the motors will react (rotation, speed,...) and the endstops will be... what thet are per default...

For a printer I modified (without knowing exactly the initial caracteristics), I proceeded motor per motor (deconnecting the others), endstop per endstop to assess first the correct behaviors of all the axes and nozzle. Then I inputed the values of the build "chamber" and calculated the correct numbers for the steps, the extrusion etc...

It is a long, minutious and  interesting task and helps a lot to understand the caracteristics of marlin. 

Good luck

fiveangle, 2017-08-01:
@bryonbaker - so did we misunderstand your suggestion, or do you think this should be closed ?

fiveangle, 2017-08-01:
I think you misunderstood the response: certainly an example could be created that takes the default Marlin code and changes only 2 characters:

`//#define REPRAP_DISCOUNT_SMART_CONTROLLER`

to 

`#define REPRAP_DISCOUNT_SMART_CONTROLLER`

But this does not help probably 99% of RAMPS users configure their printer.  The need to understand how to confiure Marlin using the existing Config*.h files is really the minimum level of entry.  If users are less savvy than that, it is not likely that they have built thier own printer so are not likely attempting to compile/install Marlin from source.

Further, even a simple search engine query of natural language "configure marlin for ramps 1.4 with lcd" comes up with a first page full of answers that are far more useful than the above change.

It's unfortunate that occasionally there is a user who struggles with the process, but all we can do is try to improve to prevent these things, and I don't see how adding another configuration example will accomplish that.

That's not to say there isn't other opportunities to improve.  Based on your experience, if you think the display section of Configuration.h needs some improvement, please submit a PR with your suggestion.  I could envision a "_These are general display features (20x4 text, and U8glib graphics) so are normally incldued automatically by selecting a specific display, so are not normally selected here_" addition, or something even simpler.  This would be far more likely to prevent users from falling into the same trap as you did than some languishing config exmample that nobody looks at, in my opinion.

There is a big desire to make configuration simpler such as @akaJes's #6722, Scott's own https://github.com/thinkyhead/MarlinConfigurator and so don't think everyone doesn't feel your pain ! :)  It's just that yet another ignored configuration example doesn't probably solve much of it :/

closed 2018-02-09 by thinkyhead