MarlinFirmware/Marlin #7345: Z axis endstop not working

RATT88, 2017-07-23:
I am trying to set up a rostock delta 3d printer. When the carriage reaches the top of the Z axis it fails to trigger the endstop switch and the motor continues to run. I have check voltages and wiring and they all is as it should be. The problem must be somewhere in the marlin firmware. I have no idea what adjustment needs to be made to the code to rectify this.
Any assistance would be greatly appreciated.

teemuatlut, 2017-07-23:
Related #7335 ?

cdedwards, 2017-07-23:
or try M119 and check the status of your switches. you might need to reverse them in configuration.h

RATT88, 2017-07-24:
#7335 describes how my printer is behaving. I have run m119 an verified that the switches are all working. Does this mean I need reverse them in configuration.h? Which piece of code do I need to alter?

ScanmanIR, 2017-07-29:
I am new to the "GitHub" forum. hopefully i am at the right location.  I Have a Hatchbox 3D Delta Style printer that has developed possibly a similar problem with what appears to be the rear Z axis end stop ( if the circuit board is labeled correctly ).  When the "Home All" is chosen the carriage raises until it hits the stops on two of the axis and the third continues to run till we shut off power.  
I believe that it is a hardware problem but I need some suggestions on how to check out the motherboard, output drivers or solidstate endstop sensors.  I am more from the old school electronics repair side than the programming side but i have some sons that can help me reload drivers if the main board output channel is dead.  

Any thoughts on diagnosis?

fiveangle, 2017-07-31:
`*_ENDSTOP_INVERTING`

closed 2018-02-09 by thinkyhead