MarlinFirmware/Marlin #7346: How Marlin interpolate trajectories

NervenCid, 2017-07-23:
Hello:

I Beg your pardon, but i'm researching about Marlin firmware for my master thesis, and i wonder how marlin makes tajectory interpolation, what algorithm uses... if it is cubic or LSBP polinomials...

Do you have recommended speeds and accelerations? i'm new on this and i still trying to research something in order to apply my limited knowledge into robotics into this field...

Thank you

Grogyan, 2017-07-23:
What area is your masters in, may I ask?

NervenCid, 2017-07-24:
Hello @Grogyan : 

I'm making my master degree in industrial automation, and i'm making my thesis about 3D printing, in fact this is my project:

http://reprap.org/wiki/UNAL_FDM

I hope to release all documentation soon, but i'm working now in theoretical work know, i'm re doing all calculus and mathematical stuff again in order to give more theorital strenght to the final report, and i'm making some simulations about speeds, and trajectories in order to calculate dynamic model.

Grogyan, 2017-07-24:
@NervenCid Those last photos you have used are from the Darwin G1 printer.  
That machine is currently in my room awaiting refurbishment.

Personally, IMO you probably need to do more physics papers.

In your OP the answer is in the Planner.cpp file

Roxy-3D, 2017-07-24:
> In your OP the answer is in the Planner.cpp file

Or from another perspective...   GRBL is the foundation for Marlin's simultanious movements.  bengler.no/grbl   

> i'm re doing all calculus and mathematical stuff again in order to give more theorital strenght to the final report, 

There is plenty of room to optimize the bed leveling area too.   And the automatic self calibration of the different axis of movement (whether Cartesian or Delta or Scara).

NervenCid, 2017-07-24:
Yep.... @Grogyan 

I'm using a template, i hope into having more time for editing the wiki, and change the pictures... the entire project is under my responsability, and i don't have time even to learn git and upload all here :(... Sadly i don't have any more time, and i even integrate web services via octoprint...

@Roxy-3D since you mention, i'm planning to use the estra stepper (i'm using r
RAMPS 1.4) in order to implement a transfer system for automatizate part extraction, i don't know if the recent version of marlin, allows to use E2 for another utilities apart of second extruder

I will upload the changes to github if i learn to use Git :(

thinkyhead, 2017-07-24:
> trajectory interpolation, what algorithm uses

Marlin uses a simple Bresenham algorithm.

closed 2017-09-26 by thinkyhead