MarlinFirmware/Marlin #7348: [FR] merge controller_fan and auto_extruder_fan functions when assigned to same pin

Dope-johnny, 2017-07-24:
Hi,
I'd like to use my last unused mosfet (E1) on my MKS 1.4 to control both the fan for the extruder AND the fan for the steppers and mosfets on the mainboard. Since both are usually running the same time my target is to use the same pin. Instead of just hooking up the controller fan to the extruder fan and use solely the extruder_auto_fan function I want to check for both - the stepper drivers and the extruder temp - in case anything unforeseen happens.

I found out that the sanity.h throws an error when defining the same pin for the controller_fan and extruder_auto_fan functions(?). I searched in the code to find anything that might cause issues if I remove this check and define both to the same pin, but I think my programming skills are too low to risk it.

So my request is to merge both functions when assigned to the same pin when either of these functions is active (or a similar workaround).

Example:
A setup like this in the configuration_adv.h
```
#define CONTROLLER_FAN_PIN HEATER_1_PIN
#define E0_AUTO_FAN_PIN HEATER_1_PIN
```

results in this behaviour:
```
If current_extruder_temp < EXTRUDER_AUTO_FAN_TEMPERATURE
  If steppers = disabled
    sleep(CONTROLLERFAN_SECS), disable(CONTROLLER_FAN_PIN)
else enable(CONTROLLER_FAN_PIN)
```

Thanks

FHeilmann, 2017-07-24:
The "problem" here is that the two things are happening in two different domains, heat and movement, which are defined in two parts of Marlin which are quite separated from each other. One would have to give the thermal manager knowledge about the steppers and the planner knowledge about the heaters which would result in a lot of info that has to be spilled across those two domains.

It is certainly doable but in the favor of clean code separation I'm not sure if it is worth it.

thinkyhead, 2018-02-09:
If connected to the same physical pin then you don't need to enable both features. When your E auto fan spins up it will simply turn on the controller fan. And you don't need to do it the other way around, so that should do the trick!

closed 2018-02-09 by thinkyhead