MarlinFirmware/Marlin #7349: Bug Report

superlegasteniker, 2017-07-24:
hello.
when enabeling lasersupport on an MKSbaseV1.2/MKSbaseV1.4 marlin used the microsteppingpin as pwm.
as a result my Xaxis did not move but made a screaming noise.
here are some changes to pins_RAMPS.h that themporaly fixed the problem:

pins_ramps
	#define SERVO1_PIN          65 //-1 //6
	#define SERVO2_PIN          66 //-1 //5
	#define SERVO3_PIN          12 //-1 //4
		//#define X_CS_PIN           53
		//#define Y_CS_PIN           49
		//#define Z_CS_PIN           40
		//#define E0_CS_PIN          42
		//#define E1_CS_PIN          44
	#define X_MS1_PIN          5
	#define X_MS2_PIN          6
	#define Y_MS1_PIN          58
	#define Y_MS2_PIN          59
	//#define Z_MS1_PIN          physical pin 78, not mapped in adruino ide
	#define Z_MS2_PIN          39
	#define E0_MS1_PIN         64
	#define E0_MS2_PIN         63
	#define E1_MS1_PIN         4
	#define E1_MS2_PIN         57
		//#if DISABLED(SDSUPPORT)
		//  #define MAX6675_SS       66 // Do not use pin 53 if there is even the remote possibility of using Display/SD card
		//#else
		//  #define MAX6675_SS       66 // Do not use pin 49 as this is tied to the switch inside the SD card socket to detect if there is an SD card present
		//#endif
	#define FILWIDTH_PIN      -1 //5   // Analog Input on AUX2
	#define FIL_RUNOUT_PIN      -1 //4
	#define PS_ON_PIN          -1 //12
#define SPINDLE_LASER_ENABLE_PIN  11 //-1 //4  // Pin should have a pullup/pulldown!
#define SPINDLE_LASER_PWM_PIN     2 //-1 //6  // MUST BE HARDWARE PWM
#define SPINDLE_DIR_PIN           65 //-1 //5

Bob-the-Kuhn, 2017-07-24:
That's not a temporary fix.  You are expected to modify pins_ramps.h to match your machine.

closed 2017-07-24 by Bob-the-Kuhn

superlegasteniker, 2017-07-24:
[pins_RAMPS.zip](https://github.com/MarlinFirmware/Marlin/files/1170889/pins_RAMPS.zip)
it is temporary, because i made those changes in ramps.h instead of mks13.h.