MarlinFirmware/Marlin #7353: G26 behaviours

gavatron3000, 2017-07-25:
Hi all,
enjoying setting up the UBL it seems so far as a powerful tool for bed leveling. bit of a learning curve too! but worth it so far.
My machine is of coreXZ configuration and im noticing during different parts of mesh editing and validation that the steppers timeout and turn off which is undesirable in my machines configuration as the Z and X axies can drift from gravity acting on them.
i guess my question is can someone else verify this or have i missed a setting, in configuration.h my stepper are set to stay on but they do timeout as per configuration_adv.h settings. 
Or are we able in G26 to keep the steppers active during the time the G26 heats up or Home after heating would be better imo. And also keep steppers active during mesh editing.

ive been able to work around this by heating up manually before mesh validation and also by proping up my z axis during fine tune all points while i edit

regards
gav

Roxy-3D, 2017-07-25:
You can bump up the #define DEFAULT_STEPPER_DEACTIVE_TIME 120  in configuration_adv.h

Or....  You can do what I do and turn it off by setting the number to 0

gavatron3000, 2017-07-25:
no dramas, thats an easy way

gavatron3000, 2017-07-25:
so for other in my situation just thought id share my changes

in configuration_adv.h i changed the following:

#define DEFAULT_STEPPER_DEACTIVE_TIME 240
#define DISABLE_INACTIVE_X false      //changed
#define DISABLE_INACTIVE_Y true
#define DISABLE_INACTIVE_Z false      //changed
#define DISABLE_INACTIVE_E true
#define HOME_AFTER_DEACTIVATE      //switched this on

X and Z axes will stay on indefinitely now, while Y and E timeout
thanks, will close now

GibaHC, 2017-07-30:
I had the same issue when the motors where turned off during heat-up. The extruder hit the bed really hard.
Maybe `G26` should home before start priming ...

Roxy-3D, 2017-07-30:
> Maybe G26 should home before start priming ...

```cpp
  /**
   * G26: Mesh Validation Pattern generation.
   *
   * Used to interactively edit UBL's Mesh by placing the
   * nozzle in a problem area and doing a G29 P4 R command.
   */
  void unified_bed_leveling::G26() {
    SERIAL_ECHOLNPGM("G26 command started.  Waiting for heater(s).");
    float tmp, start_angle, end_angle;
    int   i, xi, yi;
    mesh_index_pair location;

    // Don't allow Mesh Validation without homing first,
    // or if the parameter parsing did not go OK, abort
    if (axis_unhomed_error() || parse_G26_parameters()) return;
```

gavatron3000, 2017-07-31:
this would be great, i d prefer it disabled the timeout for the G26 "session" and restore the timer after. would this include the fine tune mesh editing "session" too?

cheers

gavatron3000, 2017-07-31:
reopen?

Roxy-3D, 2017-07-31:
> this would be great, i d prefer it disabled the timeout for the G26 "session" and restore the timer after. would this include the fine tune mesh editing "session" too?

Yes...   I guess if it makes sense to do it for the G26, it probably makes sense for the stepper motors not to time out during G29 P4 also.

gavatron3000, 2017-08-01:
perfect

closed 2017-12-16 by gavatron3000