MarlinFirmware/Marlin #7355: BUG: Controller fan PWM changes to max when entering LCD menu

JanVelleman, 2017-07-25:
### Bug Report
- Description: I use the controller fan in my printer (V1.1.4), and set the maximum PWM to 204 (80% of max) for several reasons. When printing normally this works fine, but when going into the menu structure when printing the output goes to 255 (100%) for several seconds before settling again to 204. The pin I use for this output is a hardware PWM capable pin.
- Expected behaviour: the output should stay on 204
- Actual behaviour: the output goes to 255
- Steps to reproduce:
  - Start to print with a controller fan max set to less than 255.
  - Go into menu structure on LCD to see the output go to max.

Config of the controller fan:

#define USE_CONTROLLER_FAN
#if ENABLED(USE_CONTROLLER_FAN)
  #define CONTROLLER_FAN_PIN FAN1_PIN  // Set a custom pin for the controller fan
  #define CONTROLLERFAN_SECS 1          // Duration in seconds for the fan to run after all motors are disabled
  #define CONTROLLERFAN_SPEED 204        // 255 == full speed
#endif

Tannoo, 2017-07-26:
Do you have `FAN_KICKSTART_TIME' enabled?

JanVelleman, 2017-07-27:
Strange... If I uncomment #define FAN_MIN_PWM 100 or  #define FAN_KICKSTART_TIME 5 for example (just to test) then this has no impact...

If I comment this:

      // allows digital or PWM fan output to be used (see M42 handling)
**//WRITE(CONTROLLER_FAN_PIN, speed);**
      analogWrite(CONTROLLER_FAN_PIN, speed);

in the controllerFan() function in de Marlin_main.cpp file then instead of shirtly going to max (255) when using the LCD it hos to 0... So the fan stops spinning for a bit when entering and selecting menus...

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead