MarlinFirmware/Marlin #7358: Delta homing  always skips Z_AXIS (C_AXIS) bump sequence.

jlarnal, 2017-07-25:
[configs_and_log.zip](https://github.com/MarlinFirmware/Marlin/files/1175043/configs_and_log.zip)

------------------------------------------------------------
### Bug Report
**- Description:**  G28 homes axes, but skips the GAMMA axis "bump" procedure.

**- Expected behaviour:** GAMMA (Z_AXIS, aka C_AXIS ) carriage should move aways from its endstop once reached, and reach for it again, slowly (as the other two carriages do).

**- Actual behaviour:** GAMMA carriage triggers the endstop and just sits there.

**- Steps to reproduce:** Call G28 from any position.

I've included a log from pronterface's console. The symbol DEBUG_LEVELING_FEATURE is declared and the G28 home leveling is verbose.

The "homeaxis(Z)" traces ( lines 59 to 76 of _Pronterface_log.txt_ ) are sent by the device without any physical activity from the printer.
Moreover, the M119 request macro reports triggered endstops, except for Y_AXIS (BETA tower).

getget1980, 2018-02-08:
Hello, my Delta Z axis does not bump too. It's a problem because homing is not precise on this axis. Do you have any solution ?

thinkyhead, 2018-02-17:
Have you confirmed with `M119` that the endstops are all functioning correctly with no spurious TRIGGERED states?

getget1980, 2018-02-17:
Yes, no endstops problem. When i Home with g28 Z axis does not bump. But when I do G33 it starts by a homing and all 3 axis bumps.

thinkyhead, 2018-02-27:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.