MarlinFirmware/Marlin #7360: [Q] Head crashes in X-direction when changing T0 -> T1

Kaibob2, 2017-07-26:
### Bug Report
- Description:
I'm not sure if this is a bug or if I don't get the configuration right. I have a dual extruder setup. My X axis has an overall length of 254mm. 
The endstop is mounted at the minimum (left position). The distance from the left nozzle (T0) to the bed is 32mm and the distance between the left and right nozzle is 29mm. The distance from the right nozzle (T1) to the bed is 6mm.
The travel limits are set to:

#define X_MIN_POS -32
#define Y_MIN_POS 0
#define Z_MIN_POS 0
#define X_MAX_POS 222
#define Y_MAX_POS 200
#define Z_MAX_POS 280

The problem now is, that if i home all axes (LCD shows X=-32) and i have selected T0 as active extruder the gantry tries to move 29mm to the left, but there is no space left so it crashes into the left carrier block. (I currently avoid this behaviour by enabling  `#define ENDSTOPS_ALWAYS_ON_DEFAULT` but this is not an option anymore because i want to use `#define MESH_BED_LEVELING`)
If T1 is selected before homing and afterwards i select T0 as active extruder the X-Gantry moves 29mm to the right but the LCD still shows X=-32.

This is the configuration: [K3DV16.zip](https://github.com/MarlinFirmware/Marlin/files/1176214/K3DV16.zip)

The question is: How does this setup have to be configured? I tried each and every combination but i don't get it.

[image]

[image]

kcheeeung, 2017-07-28:
Try this.

What is `#define USE_ZMAX_PLUG` used for?

**make this all TRUE**
```
// Mechanical endstop with COM to ground and NC to Signal uses "false" here (most common setup).
```

**comment this out**
`//#define ENDSTOP_INTERRUPTS_FEATURE `

Kaibob2, 2017-07-28:
` #define USE_ZMAX_PLUG` is for an endstop mounted at the very end of the Z-Axis to prevent a crash there.

> make everything all TRUE

This makes no sense. I use optical endstops and they are low-active. If i make everything true they will be considered triggered the whole time.

> comment this out
//#define ENDSTOP_INTERRUPTS_FEATURE

Why??

Kaibob2, 2017-08-03:
I got it working, but to be honest I don't exactly know how :)
Maybe it was an operator error. Sorry.
TWIMC: This is the working config
[Working.zip](https://github.com/MarlinFirmware/Marlin/files/1196240/Working.zip)

closed 2017-08-03 by Kaibob2