MarlinFirmware/Marlin #7361: Manual Mesh Bed Leveling Not Saving pts??

Moostemaster, 2017-07-26:
Hey everyone, i am completely new to 3d printing and finished assembling by repRapGuru Prusa i3 mk2 about a week ago. At first using repRapGuru's version of Marlin, things where printing fine without too much issue but after i was continually having bed leveling problems, i decided to upgrade to the true marlin firmware and look into the manual mesh bed leveling via my lcd panel. Although i can access the option through the panel and begin manually probing, when i look at the feedback through repetier host, it seems to just be reading back all points values as just being zero and ultimately causing no change to my prints. So basically it would read back x:0 Y:0 Z:0 instead of the point value and the individual z offset for each point. So does anyone know what could be the possible issues or if i happen to be doing things wrong? Any help is greatly appreciated, i just want to finally have a functional 3d printer.

Moostemaster, 2017-07-26:
I would to give manual mesh bed leveling a try via g-code instead of through my lcd panel as a possible fix but i dont know the actual code to do it, if someone can enlighten me that would be great as well. many thanks in advance

fiveangle, 2017-07-27:
https://github.com/MarlinFirmware/Marlin/wiki/Mesh-Bed-Leveling

chanddan, 2017-11-02:
@fiveangle isn't mesh bed levelling default in the easyconfig? Not sure how this helps?

closed 2018-02-09 by thinkyhead