MarlinFirmware/Marlin #7362: UBL Question

gcormier, 2017-07-26:
When starting a print, do we need to issue any sort of GCode to load the mesh? Or once it's set, it's applied always unless disabled?

Right now (old) startup code would be G28; G29;

Should the new one just be G28? Or maybe, G28;G29 A; ?

Roxy-3D, 2017-07-26:
The answer depends on how you have UBL configured.   If you save (do a M500) the printer's state with UBL active and a mesh loaded from EEPROM...   It will do that when the printer wakes up after a power cycle or reset.

If you saved your state with UBL disabled...   You will need to do what ever G29 commands you desire prior the print really starting.    This split personality was to give people the option of controlling whether G29 was automatically active or not.

I don't even have a G29 in my Slicer's startup gcode.  UBL is alive and functioning when my printer powers up.

gcormier, 2017-07-26:
Awesome. I'll do that. So then the only thing for startup is G28?

So far so good for my slightly domed PEI surface :)

Roxy-3D, 2017-07-26:
Yes...  Just a G28 and what ever else you do to pre-heat the nozzle and bed...

closed 2017-07-26 by gcormier