MarlinFirmware/Marlin #7364: UBL Mesh doesn't account for Z probe offset

naviathan, 2017-07-27:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

### Bug Report
- Description: When using UBL on a Cartesian printer the mesh doesn't account for the Z probe offsets while probing the plate. During the mesh generation the probe moved off the plate causing the nozzle to crash into the build platform. 
- Expected behaviour: UBL Mesh Generation should account for probe offset to keep the probe on the build platform while generating the mesh.
- Actual behaviour: UBL doesn't account for probe offset and nozzle crashes into the build platform when the mesh generation moves to the outer edges of the plate.
- Steps to reproduce: Install a Z probe with an offset from the nozzle. Start UBL Mesh Generation
  
Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

****Please note I have added the probe offset in my configuration_adv.h. It's a crude implementation that doesn't account for probes with a positive offset or any variations beyond my personal configs. It does work though. ****

[Marlin Configs.zip](https://github.com/MarlinFirmware/Marlin/files/1180394/Marlin.Configs.zip)

Roxy-3D, 2017-07-27:
The whole reason the automatic probing can't probe all mesh points is because it does know about the Z-Probe's X & Y offset from the nozzle.   Where (what coordinates) is the nozzle going off the bed?

naviathan, 2017-07-27:
For me (because my offset is a negative one) the probe goes off the bed to the left on the outer most points. I don't know what coordinates exactly. In the configuration_adv.h it accounts for the inset in the limits:

```
#define UBL_MESH_INSET 10
#define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
#define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
#define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
#define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET))
```

but not the probe offset. I added the probe offset myself (this will only work with a negative offset):

```
#elif ENABLED(AUTO_BED_LEVELING_UBL) // Added probe offsets to UBL mesh measurements.
  #define UBL_MESH_MIN_X (X_MIN_POS - X_PROBE_OFFSET_FROM_EXTRUDER + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER - UBL_MESH_INSET)
  #define UBL_MESH_MIN_Y (Y_MIN_POS - Y_PROBE_OFFSET_FROM_EXTRUDER + UBL_MESH_INSET)
  #define UBL_MESH_MAX_Y (Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER - UBL_MESH_INSET)
```

With those changes my nozzle stays on the bed within the inset.

Roxy-3D, 2017-07-27:
The Z-Probe's X & Y offset are factored into the decision.   The automatic probing calls find_closest_mesh_point_of_type() and these lines insure the probe can reach the location:

https://github.com/MarlinFirmware/Marlin/blob/1.1.x/Marlin/ubl_G29.cpp#L1402-L1408

Can you add some SERIAL_ECHO() lines around that code and see what it thinks is happening?   I checked the macros that line is using and they look correct.  And they do factor in the X & Y offset.

So...   I'm a little bit confused...

naviathan, 2017-07-27:
> 1.x/Marlin/ubl_G29.cpp#L1402-L1408

I can try that tonight at some point.

Roxy-3D, 2017-07-27:
Also...   Before adding the debug SERIAL_ECHO() lines...  You do want to change this back
```cpp
  #define UBL_MESH_MIN_X (X_MIN_POS - X_PROBE_OFFSET_FROM_EXTRUDER + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_MAX_POS + X_PROBE_OFFSET_FROM_EXTRUDER - UBL_MESH_INSET)
  #define UBL_MESH_MIN_Y (Y_MIN_POS - Y_PROBE_OFFSET_FROM_EXTRUDER + UBL_MESH_INSET)
  #define UBL_MESH_MAX_Y (Y_MAX_POS + Y_PROBE_OFFSET_FROM_EXTRUDER - UBL_MESH_INSET)
```
The problem with these modifications is you are going to give up a fair amount of your 'Print Envelope' doing this.    You are literally making the bed much smaller than it has to be.

naviathan, 2017-07-27:
Interestingly enough it appears to scan a large majority of the bed like
this.

Roxy-3D, 2017-07-27:
> Interestingly enough it appears to scan a large majority of the bed like this.

This doesn't seem right.   Can you also turn of the UBL Bed Leveling (G29 D) and move the nozzle to the four edges of the bed?    We need to verify that X=0 and Y=0 is at the left and front edge of the bed.   And then do the similar thing for the right edge and back edge of the bed.   

There is something else going on here.

naviathan, 2017-07-27:
Sure, I'll double check that tonight.  When I auto home, my X carriage
pulls left to the min stop, the y carriage pulls back to the min stop
(leaving the nozzle over the front left corner of the bed) then the bed
comes forward and the X carriage pulls right to home the Z probe on the
center of the bed. I would assume, given the printer's homing actions, that
the left front corner is indeed 0,0. I used bilinear prior to this and it
mapped the bed just fine with the probe. Switched to UBL out of curiosity
and you know the rest.

Roxy-3D, 2017-07-27:
Thanks!    That will be both helpful and appreciated.

naviathan, 2017-07-27:
Here's what I added:
....
// If using the probe as the reference there are some unreachable locations.
          // Also for round beds, there are grid points outside the bed the
nozzle can't reach.
          // Prune them from the list and ignore them till the next Phase
(manual nozzle probing).
          SERIAL_ECHOLN(mx);
          SERIAL_ECHOLN(my);
          SERIAL_ECHOLN(position_is_reachable_by_probe_raw_xy(mx, my));
          SERIAL_ECHOLN(position_is_reachable_raw_xy(mx, my));
          if (probe_as_reference ?
!position_is_reachable_by_probe_raw_xy(mx, my) :
!position_is_reachable_raw_xy(mx, my))
            continue;
....

I also changed the other section back to its original. I don't know what is
going on but it didn't hit the bed this time. I got this output:

....
echo:busy: processing
echo:busy: processing
10.00
186.89
0
1
10.00
209.00
0
1
32.11
186.89
0
1
32.11
209.00
0
1
54.22
186.89
0
1
54.22
209.00
0
1
76.33
186.89
0
1
76.33
209.00
0
1
98.44
186.89
0
1
98.44
209.00
0
1
120.56
186.89
0
1
120.56
209.00
0
1
142.67
186.89
0
1
142.67
209.00
0
1
164.78
186.89
0
1
164.78
209.00
0
1
186.89
186.89
0
1
186.89
209.00
0
1
209.00
10.00
0
1
209.00
32.11
0
1
209.00
54.22
0
1
209.00
76.33
0
1
209.00
98.44
0
1
209.00
120.56
0
1
209.00
142.67
0
1
209.00
164.78
0
1
209.00
186.89
0
1
209.00
209.00
0
1
echo:busy: processing
echo:busy: processing
....

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Roxy-3D, 2017-07-27:
That is very strange....  It is claiming the location is never position_is_reachable_by_probe_raw_xy(mx, my));
That doesn't make any sense ????

naviathan, 2017-07-28:
I was thinking the same thing.

naviathan, 2017-07-28:
Could it have something to do with the Z probe offsets being negative?

Roxy-3D, 2017-07-28:
I don't think so...   In fact, most of my Z-Probe offsets are negative.   And here is the other thing:  If the Z-Probe offsets were not being respected, it wouldn't be just you complaining about them.   That code is some of the most mature code in the UBL system.   (Just because it is very basic and needed to be in place for everything.)

It might be worth while starting fresh with a new copy of https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x/Marlin if that will compile in Arduino.  Otherwise, start fresh with:  https://github.com/MarlinFirmware/Marlin/tree/bugfix-1.1.x/Marlin

Something else is entering into the equation.

naviathan, 2017-07-28:
> MarlinFirmware/Marlin/tree/bugfix-1.1.x/Marlin

I will give that a go. So far so good tonight. I haven't had any issues.

Redferne, 2017-07-30:
I have a similar issue with the following configuration:

```
// Travel limits after homing (units are in mm)
#define X_MIN_POS -41
#define Y_MIN_POS -3
#define Z_MIN_POS 0
#define X_MAX_POS 220
#define Y_MAX_POS 220
#define Z_MAX_POS 240

#define X_PROBE_OFFSET_FROM_EXTRUDER 44  // X offset: -left  +right  [of the nozzle]
#define Y_PROBE_OFFSET_FROM_EXTRUDER 15  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER 0  //+4mm glass // Z offset: -below +above  [the nozzle]

  #define UBL_MESH_INSET 35          // Mesh inset margin on print area
  #define GRID_MAX_POINTS_X 5 // 10      // Don't use more than 15 points per axis, implementation limited.
  #define GRID_MAX_POINTS_Y GRID_MAX_POINTS_X
```

After running G28, G29 P0, G29 P1 and G29 W says the following:

```
Send: G29 W
Recv: Unified Bed Leveling System v1.01 inactive.
Recv: Mesh 0 Loaded.
Recv: UBL object count: 1
Recv: planner.z_fade_height : 10.0000
Recv: zprobe_zoffset: 0.0000000
Recv: UBL_MESH_MIN_X  (-41 + 35)=-6
Recv: UBL_MESH_MIN_Y  (-3 + 35)=32
Recv: UBL_MESH_MAX_X  (220 - (35))=185
Recv: UBL_MESH_MAX_Y  (220 - (35))=185
Recv: GRID_MAX_POINTS_X  5
Recv: GRID_MAX_POINTS_Y  5
Recv: MESH_X_DIST  47.75
Recv: MESH_Y_DIST  38.25
Recv: X-Axis Mesh Points at: -6.000  41.750  89.500  137.250  185.000
Recv: Y-Axis Mesh Points at: 32.000  70.250  108.500  146.750  185.000
Recv: Kill pin on :41  state:1
Recv: 
Recv: ubl_state_at_invocation :0
Recv: 
Recv: ubl_state_recursion_chk :0
Recv: 
Recv: Meshes go from 0x0268 to 0x0F7F
Recv: sizeof(ubl) :  1
Recv: 
Recv: z_value[][] size: 100
Recv: 
Recv: EEPROM free for UBL: 0x0D17
Recv: EEPROM can hold 33 meshes.
Recv: 
Recv: Unified Bed Leveling sanity checks passed.

```
Topo:

```
Send: G29 T0
Recv: 
Recv: Bed Topography Report:
Recv: 
Recv: (0,4)                           (4,4)
Recv: (-6,185)                        (185,185)
Recv:      .       2.222    2.370    2.452    2.532
Recv: 
Recv:      .       2.148    2.092    2.345    2.375
Recv: 
Recv:      .       1.970  [ 2.202]   2.392    2.325
Recv: 
Recv:      .       2.040    2.087    2.247    2.385
Recv: 
Recv:      .       2.013    2.172    2.285    2.305
Recv: (-6,32)                            (185,32)
Recv: (0,0)                             (4,0)

Send: G29 T1
Recv: 
Recv: Bed Topography Report for CSV:
Recv: 
Recv: NAN,2.222,2.370,2.452,2.532
Recv: NAN,2.148,2.092,2.345,2.375
Recv: NAN,1.970,2.202,2.392,2.325
Recv: NAN,2.040,2.087,2.247,2.385
Recv: NAN,2.013,2.172,2.285,2.305
```

UBL cannot reach UBL_MESH_MIN_X since it is negative -6? This seems weird. If I order nozzle to 0,0 it sits nicely at the edge of the bed. How is UBL calculating the actual probe position? It only seems to use *MIN_POS and UBL_MESH_INSET. Should it even use MIN_POS, since we are only interested in the printable area, right? I know my INSETs are high, but I have some flimsy clips holding the glass in place, reducing the printable area.

And as a follow up question. How to properly set the Z offset based on the topo above? Since the Z min is now at Z pos 0 (G28), but UBL probe reads 2.202 at center of the bed, where it should read something very close to 0.

Please help me to get these numbers in order :sob:

Roxy-3D, 2017-07-30:
> And as a follow up question. How to properly set the Z offset based on the topo above? Since the Z min is now at Z pos 0 (G28), but UBL probe reads 2.202 at center of the bed, where it should read something very close to 0.

The 'best practice' is to do the Z measurement for homing at the center of the bed.    And make sure you can print a small object at the center of the bed without any bed leveling active.    With that working, if you build a mesh, the numbers should mostly be very close to 0.00.

Redferne, 2017-07-31:
Yes. The G28 homing correctly sets Z0 with the nozzle nudging the bed. A small print without UBL works nicely. But trying to do UBL with G29 P1 gives me the topo above. So there's an offset of approx 2.2mm from ZMIN limit (Z0) and the BLTouch measurement. Do I physically move the BLTouch or use M851? Sorry for being daft :cry:

Roxy-3D, 2017-07-31:
> So there's an offset of approx 2.2mm from ZMIN limit (Z0) and the BLTouch measurement. Do I physically move the BLTouch or use M851? Sorry for being daft :cry:

I set the Z_PROBE_OFFSET_FROM_EXTRUDER so I can print something very small.    And then I bump that number I tiny bit up or down to get it to print the first layer (of a small object) perfect.    And then I build a mesh and edit it.   That works really well...    On my gMax there is a BL-Touch probe.   I've got raised up as much as I dare...  It is at:  https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/example_configurations/gCreate/gMax1.5%2B/Configuration.h#L679

Redferne, 2017-07-31:
Thanks. Just found the missing piece of the puzzle here [M851](http://marlinfw.org/docs/gcode/M851.html) 

Any idea on setting the correct bed information in my original question? My bed is 220x220 and the nozzle X carriage hits limit at -41. I have clips holding the glass on the heated bed reducing the printable area to 150x150 (35 UBL Inset). 
However using the above configuration, UBL cannot probe the leftmost column at X pos -6. Whats the suggested way of defining the bed? It seems like UBL doesnt allow negative X and/or doesn't account for X probe offset?

I know you guys are busy busy with the 32bit migration, thanks for your time!

gavatron3000, 2017-07-31:
i found on my setup:

-z probe is used as an endstop - used its offset -1.9mm in my machine as a manual z home pos of 1.9mm
-set z probe offset to 0.00mm
-z safe homing must be defined
-travel limits must be only the bed size
- disable software endstops after homing if you need to move beyond the bed area

this work reliably in my case anyways
cheers

Roxy-3D, 2017-07-31:
> However using the above configuration, UBL cannot probe the leftmost column at X pos -6. Whats the suggested way of defining the bed? It seems like UBL doesnt allow negative X and/or doesn't account for X probe offset?

Take a look at the gCreate/gMax example configuration_adv.h  file.     That is for my printer.    It has a VERY large bed.   I could not get a silicon heatpad and piece of borosilicate glass for the full size.      So...   I have 400 x 400 mm in the center of the bed.

I just adjusted the min and max positions along with the inset such that the machine homes X & Y to a very negative number...    And the left front of where my glass is on the bed is (0,0)

benlye, 2017-08-01:
I'm just starting to set up UBL, and my printer also has a negative X_MIN_POS, and an X_MAX_POS where the nozzle is off the bed.  With the new BED_SIZE parameters properly configured I found that this worked well to get the UBL mesh properly positioned on the bed:

```
  #define UBL_MESH_MIN_X (0 + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y (0 + UBL_MESH_INSET)
  #define UBL_MESH_MAX_Y (Y_BED_SIZE - (UBL_MESH_INSET))
```

Roxy-3D, 2017-08-01:
Yeah...  But the problem is if your X-Min or Y-Min is not at 0.0    I don't think this is a valid use of X_BED_SIZE either:
```cpp
//#define Z_SAFE_HOMING

#if ENABLED(Z_SAFE_HOMING)
  #define Z_SAFE_HOMING_X_POINT ((X_BED_SIZE) / 2)    // X point for Z homing when homing all axis (G28).
  #define Z_SAFE_HOMING_Y_POINT ((Y_BED_SIZE) / 2)    // Y point for Z homing when homing all axis (G28).
#endif
```
I think we should be using X_MIN_POS instead!   But...  so be it...

benlye, 2017-08-01:
> Yeah... But the problem is if your X-Min or Y-Min is not at 0.0... 

That's exactly the point - neither my x-min nor y-min is at 0, they are both negative, so using those for any positioning doesn't work too well. The minimum edge of the bed however is 0 (I know this isn't true for everyone), however, and the max edge is bed_size, not max_pos, because the carriage can move past the edge. 

Each to their own; at least it's easily configured now :-)

closed 2018-02-09 by thinkyhead