MarlinFirmware/Marlin #7369: Marlin 1.1.4 PROBING_FANS_OFF

azx987sa, 2017-07-28:
Good day to all. A remarkable addition is #define PROBING_FANS_OFF, which should turn off coolers during table calibration. But only the blow-off fan of the part is turned off, which by definition must be turned off before the printing of the specified layer begins.
I can not turn off the hot-air blower cooler. The pin of this cooler is configured in Configuration_adv.h #define E0_AUTO_FAN_PIN 5. In the temperature range on / off as expected. Need to make additional changes to the configuration.h configuration_adv.h?

tcm0116, 2017-07-28:
The extruder fans are not currently disabled by PROBING_FANS_OFF. An alternate approach you could take is to perform the G29 command before you heat the extruder, since the extruder fan should be off while it is cold.

Roxy-3D, 2017-07-28:
I think it can be argued that all of the fans are potential electrical noise sources during the probing.   Shouldn't we turn all fans off during the actual probe event if this option is enabled?

tcm0116, 2017-07-28:
@Roxy-3D we should definitely turn off the extruder fans, so long as they're only turned off in conjunction with the extruder heaters.

thinkyhead, 2018-02-09:
Typically the fan blowing on the hotend fins is tied to a 12V always-on pin. We can't control it, so it stays on. The fans we can control are turned off.

closed 2018-02-09 by thinkyhead