MarlinFirmware/Marlin #7370: on my Rambo v1.3 board when i comment out #define AUTO_BED_LEVELING_BILINEAR i get the following error, any ideas ?  pasting "SWITCH_ENABLED_" and ">>" does not give a valid preprocessing token

stefi01, 2017-07-28:
Thank you for submitting your feedback to the Marlin project.
Please use one of the templates below to fill out this box.

------------------------------------------------------------
### Feature Request
Please put [FR] in the issue title: `[FR] Add-on that goes 'ping'`

------------------------------------------------------------
### Compile Error
When I compile with `FEATURE_X` I get an error:
```
Paste_the_error_text_here
```

------------------------------------------------------------
### Bug Report
- Description: ---
- Expected behaviour: ---
- Actual behaviour: ---
- Steps to reproduce:
  - Do this
  - Do that

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

stefi01, 2017-07-28:
[Configuration_adv.txt](https://github.com/MarlinFirmware/Marlin/files/1182813/Configuration_adv.txt)
[Configuration.txt](https://github.com/MarlinFirmware/Marlin/files/1182814/Configuration.txt)

stefi01, 2017-07-28:
files uploaded

stefi01, 2017-07-28:
fixed it, no worries

closed 2017-07-28 by stefi01

amacgregor, 2017-10-14:
What was it ?

stefi01, 2017-10-17:
i cant remember even though i should have

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>