MarlinFirmware/Marlin #7373: 'ADC_KEYPAD_PIN' was not declared in this scope. Very confused by this little help?

Tylerun, 2017-07-28:
### Compile Error
When I compile with `` ANET_KEYPAD_LCD" I get an error when compiling on a melzi 2.0 board. ADC_KEYPAD_PIN was not declared in this scope. Not to good at parusing Github so might have missed someone with the same problem sorry if i did. Little help or direction would be nice. 

Paste_the_error_text_here
C:\Users\~\AppData\Local\Temp\arduino_build_14835\sketch\temperature.cpp: In static member function 'static void Temperature::isr()':

temperature.cpp:2013: error: 'ADC_KEYPAD_PIN' was not declared in this scope

         START_ADC(ADC_KEYPAD_PIN);

                   ^

C:\Users\~\AppData\Local\Temp\arduino_build_14835\sketch\temperature.cpp:1921:55: note: in definition of macro 'SET_ADMUX_ADCSRA'

   #define SET_ADMUX_ADCSRA(pin) ADMUX = _BV(REFS0) | (pin & 0x07); SBI(ADCSRA, ADSC)

                                                       ^

C:\Users\~\AppData\Local\Temp\arduino_build_14835\sketch\temperature.cpp:2013:9: note: in expansion of macro 'START_ADC'

         START_ADC(ADC_KEYPAD_PIN);

         ^

Using library LiquidCrystal at version 1.0.5 in folder: C:\Program Files (x86)\Arduino\libraries\LiquidCrystal 
exit status 1
'ADC_KEYPAD_PIN' was not declared in this scope

benlye, 2017-07-28:
You need to edit your pins file to define a pin to be used as ADC_KEYPAD_PIN.  Not sure what it will be on the Melzi (someone who knows may come along shortly).  Take a look in the ANET10 pins file if you want an example.

Tylerun, 2017-07-28:
Don't much know what im doing so looks like thats staying disabled...

Tylerun, 2017-07-28:
http://prntscr.com/g1iggy

This is my pin layout i just dont know how to input that into the sanguino 11 pins

Tylerun, 2017-07-28:
Nevermind I just learned my pad works with resistance settings. I have no idea what firmware this dang thing originally came with but I dont know any that support resistance.

Tylerun, 2017-07-28:
http://prntscr.com/g1iky2 If any of you want to look into supporting it here are the pins

benlye, 2017-07-28:
> Nevermind I just learned my pad works with resistance settings. I have no idea what firmware this dang thing originally came with but I dont know any that support resistance.

That's ADC AFAIK.  The buttons are connected to the pin via resistors, with each button have a different resistance so altering the value on the pin.

Looking back to, in #6984 @ZirakC posted a configuration example for his Melzi where he used pin 1 for ADC.

What is your printer?  Which board do you have declared, and which pins file are you using?

Tylerun, 2017-07-28:
Prusa i3. Melzi 2.0. And its using Sanguino 11 as the pins.

Tylerun, 2017-07-28:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1183874/Marlin.zip)
Ill bring in my config files just in case. I dont think I am crazy but thats what it appears to be doing. The only thing not working is the lcd so i disabled it.

Tylerun, 2017-07-28:
DUDE I LOVE you for linking me that. I grabbed his source code and edited it to work on mine. OH MY GOD IT FINALLY WORKS. <3 MAJOR LOVE TO YOU and HIM

closed 2017-07-28 by Tylerun

benlye, 2017-07-28:
If I remember correctly, you can enable PINS_DEBUGGING and use M43 to check the pin values. Just hold down a button and send M43 on a serial console to see which pin shows the ADC value. You might need to do it a few times to see the change. 

Is the LCD's display working?