MarlinFirmware/Marlin #7374: enqueue_and_echo_commands_P calls drain_injected_commands_P only once.

marcio-ao, 2017-07-28:
When using "enqueue_and_echo_commands_P", it is possible to enqueue multiple commands by separating them with '\n'. However, enqueue_and_echo_commands_P() only calls drain_injected_commands_P() once, so only the first command gets executed in immediately, the others do not.

Since drain_injected_commands_P() returns true if more commands remain, would it make sense to make a loop, i.e:

void enqueue_and_echo_commands_P(const char * const pgcode) {
  injected_commands_P = pgcode;
  while(drain_injected_commands_P()); // execute commands asap (when possible)
}

My use case is that I want to be able to execute a sequence of GCODE commands after G28 completes, and so I added a  enqueue_and_echo_commands_P("G92 Z5.5\nG0 Z16") to the end of gcode_G28. However, what happens is that G92 Z5.5 happens immediately, but G0 Z16 only happens after a few other commands from the buffer are executed. I believe the above loop would fix the issue, but I do not know whether it would break anything else.

thinkyhead, 2018-02-09:
That would work.

thinkyhead, 2018-02-09:
It's vital to also break out of the `while` loop in the event that the queue is full. Otherwise it will never return.

marcio-ao, 2018-08-20:
Closing this because it sounds like the current behavior is there for a reason. Also, execute_commands_immediate_P introduced in PR https://github.com/MarlinFirmware/Marlin/pull/10450 is a far better alternative to enqueue_and_echo_commands_P.

closed 2018-08-20 by marcio-ao