MarlinFirmware/Marlin #7375: bugfix-2.0.x doesnt build with arduino ide for avr

alexxy, 2017-07-29:
Hi!

Does bugfix-2.0.x branch supposed to build for avr?

 I get error

In file included from /tmp/arduino_build_688061/sketch/pins.h:642:0,
                 from /tmp/arduino_build_688061/sketch/MarlinConfig.h:33,
                 from /home/alexxy/Develop/3D/Marlin/Marlin/Marlin.ino:31:
/tmp/arduino_build_688061/sketch/src/HAL/HAL_spi_pins.h:24:26: fatal error: MarlinConfig.h: No such file or directory
 #include "MarlinConfig.h"

teemuatlut, 2017-07-29:
For now you can successfully build it on PlatformIO.
But Arduino IDE compatibility is a priority as that is what most people use.

bobc, 2017-07-29:
It builds with Arduino 1.6.8 with 2 changes:

1. In HAL_spi_pins.h
-#include "MarlinConfig.h"
+#include "../../MarlinConfig.h"

2. Delete or move HAL_LPC1768 so it is not in Marlin/...

Tannoo, 2017-07-29:
Isn't it "..\..\MarlinConfig.h"?

bobc, 2017-07-29:
https://stackoverflow.com/questions/5790161/is-the-backslash-acceptable-in-c-and-c-include-directives

fiveangle, 2017-07-31:
Actually, it requires more changes than that (not sure as I've not got it working yet - had to stop on the 3rd .h file path fix - will try to look at later).

bobc, 2017-07-31:
I built the code using the default config, but there is SPI support that needs fixing, Chris is working on that. I would wait for him.

What areas are you finding problems?

fiveangle, 2017-07-31:
so far, just more header path issues beyond MarlinConfig.h.  Mostly stuff in the LPC HAL tree (there's also minor cleanup that i was doing as i was fixing - some directories using all lowercase, some first-letter capitalized, etc)

bobc, 2017-07-31:
If you are building for AVR, it should not be including anything in LPC HAL...  one of my suggested changes was to remove that folder!

p3p, 2017-07-31:
@fiveangle I wouldn't do cosmetic changes to things just yet some parts will not be around for ever, 

Other than SPI missing from other platforms there shouldn't be any compile errors on my development branch, the Marlin bugfix-2.0.x is behind a few commits, I need to get a PR in that will throw errors for incomplete HAL implementation (I'm having issues with trying to do so much in the preprocessor for the AVR platform)

> one of my suggested changes was to remove that folder!

That's done Bob, I've created a frameworks folder level with src for any platform specific framework code that Arduino IDE should ignore.

fiveangle, 2017-08-01:
so we are supposed to "hurry up and wait" for someone else to fix stuff ? ;)

p3p, 2017-08-01:
Well, you don't have to wait but things are still being refactored in the HAL and in base 2.0.x and any PR may be hard to merge with those overarching changes without close cooperation, once 2.0.x is actually "layout complete" it will be clearer what is actually staying around and needs cleaned up.

fiveangle, 2017-10-09:
Builds now (although you may hit command-length limit in Windows, so need to install debug arduino-builder.exe available here:

https://github.com/arduino/arduino-builder/pull/249#issuecomment-332112632

closed 2017-11-15 by thinkyhead