MarlinFirmware/Marlin #7376: Potential bug: The new print area definition seems to introduce a problem on Z axe

Bergerac56, 2017-07-29:
### Bug Report
- Description: The last bugfix version (Latest commit 8b0bd44 ), introduces an evolution of the definition of the print area. After having compared (as carefully as I can...) the configuration.h and configuration_adv.h files and added the new modifications introduced by this last version, I tried to print a test cube, not far from the limits of the bed (Bed= 310x210. Printed cube 270x190x8)

- Expected behaviour: With all previous versions of bugfix, the cube is correctly printed.

- Actual behaviour: Starting from the corner of the 0,0 (home) position, the printing head follows the X axe (270mm) and "turns" to start the Y "line" till the mid of this line. Then it stops and,  immediately, moves  (increases) the Z axe. Nothing stops it (go beyond the Z_Max value). You need to do a urgent stop to avoid going beyound the physical limits of the printer.

- Steps to reproduce:
In the zip file, I join the files "before" and "after" the recent modifications of the last bugfix version. The files "before" were used with the previous version of Bugfix. The files "After" with the last version.

To reproduce the bug, you need to print a cube, near the limits of the printing area. With a small enough cube, the problem does not occur. The max limit of the X side seems to be around 260mm

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.
[Jacbot OdB.zip](https://github.com/MarlinFirmware/Marlin/files/1185042/Jacbot.OdB.zip)

fiveangle, 2017-07-31:
You used to have travel limits set to 310 x 210 and in new config, you now have bed size set to 200 x 200 and travel limtis set to same, so to summarize the bug description:

"_**Attempting to print beyond the set physical bed size results in Z-axis raising indefinitely**_" ?

It sertainly shouldn't rise on it's own in this scnario and certainly not rise indefinitely.  First, let's confirm the claimed failure of `#define MAX_SOFTWARE_ENDSTOPS`  on the Z axis (without running your Z off the end):

 Can you move print head close to bed then change FW from what you have now `#define Z_MAX_POS 200` to `100` then rerun your repro steps and verify that when the problem occurs, the Z does absolutely rise above 100 on it's own ?

FHeilmann, 2017-07-31:
UBL is known to have this issue when you attempt to print outside the edges of the probed mesh.

Between that and what @fiveangle already pointed out that's where your problems come from.

fiveangle, 2017-07-31:
> UBL is known to have this issue when you attempt to print outside the edges of the probed mesh.

Didn't know that Florian !

Is that known accepted behavior of UBL, or known issue that is being fixed ?

FHeilmann, 2017-07-31:
It somewhere on the to-do list of @Roxy-3D I believe. For now the mantra "Don't print where you didn't probe" seems to be in order.

fiveangle, 2017-07-31:
Seems like could be a bit disasterous for anyone with bed > 200 square who miss the new bed size config since every one of thm will be running their Z off the end stops.

Bergerac56, 2017-07-31:
Right. It could be disastrous.

 BTW, I made a mistake in the file "after" I sent. Yes, the bed is 200*200 in this file, but it is a mistake. The file I used for test is well 310*210. In fact, I tested a lot of parameters to understand what was happening and rebuilt a configuration.h as clean as I could to include in the zip. I forgot to modify the size...

"For the fun", I did the test you asked and confirm that Z is going forever...

I did another test: Create a print file WITHOUT activating UBL. And it works fine. No strange Z behavior. This seems to confirm what Florian was saying.

fiveangle, 2017-07-31:
> This seems to confirm what Florian was saying.

Florian said that attempting to print outside of print area when UBL is enabled will result in Z rising.  You claim your bed size is set correctly and you are not attempting to print outside of it, so they appear very different.

@Roxy-3D - given the high likelihood of users not catching the requirement to set their bed dimensions in Configuration.h when upgrading, do you think the Z-rises-indefinitely issue Florian mentioned above really needs to be prioritized for resolution ?

Roxy-3D, 2017-07-31:
> given the high likelihood of users not catching the requirement to set their bed dimensions in Configuration.h when upgrading, do you think the Z-rises-indefinitely issue Florian mentioned above really needs to be prioritized for resolution ?

Yes.   And it has been on the 'To Do' list too long.    I was looking at it in more detail when you posted.  The problem is I was going to put the patches in bugfix-2.0.x but that doesn't compile clean right now.   And certainly it won't pass the Travis tests.     

But I'm not ready to do the change yet anyway.  Hopefully the Travis stuff will be cleaned up by the time I have the fixes.   I also will probably sneak in the changes to keep the Stepper motors from timing out during the bed & nozzle heatup in G26 and if the user lets a G29 P4 R sit idle for a long time.

Roxy-3D, 2017-07-31:
@Bergerac56    Did you mean to enable   #define MIN_SOFTWARE_ENDSTOPS   ???   For the mesh bed leveling systems, this tends to be a bad thing because the nozzle can't track the bed if there is a dip.

But with that said....   I'm having trouble duplicating what you are seeing.   Can you do a few G1 X??.?? Y??.?? commands and isolate exactly when it happens.    Maybe move the nozzle to X=100 & Y=198  and then attempt to move a short distance (off the bed) and make it happen.

fiveangle, 2017-07-31:
> I was going to put the patches in bugfix-2.0.x but that doesn't compile clean right now.

Is fixing in bugfix-2.0 important ?  Seems best to target fixing in bugfix-1.1.0 since that's where the issue exists today and where everyone who might hit the config change trigger will encounter it.

Roxy-3D, 2017-07-31:
I guess that makes some sense.    But...   I'm having trouble duplicating the failure.

fiveangle, 2017-07-31:
> I'm having trouble duplicating the failure.

Florian said above this was a longstanding issue.  Perhaps it got resolved by proxy somewhere along the lines ?

@FHeilmann - is there another issue tracking the "known issue" of Z climbing when trying to go outside the print area with UBL enabled that you mentioned above ?

Roxy-3D, 2017-07-31:
The way the UBL code works is it interpolates the appropriate Z correction from the 4 corners of the mesh cell.    If a corner does not have a value, it will be NAN (not a number).   That value is used anyway because the NAN value will propagate to the final value.   The final correction value is checked against NAN and is set to 0.000 if that is the case.

I've checked everywhere the Z-height correction is calculated, and that check against NAN is always done.   So...   I being able to duplicate the issue would be very helpful.

> Perhaps it got resolved by proxy somewhere along the lines ?

This is possible because oldmcg did some nice fixes and improvements when he made UBL work with Delta Printers.  But @Bergerac56 made it sound like he is using the latest bugfix-1.1.x so he should have anything that oldmcg did.

FHeilmann, 2017-08-01:
I admit that I am talking a little out of my ass with the "known issue" thing, in a sense that I've read it a couple of times before, with going outside of the probed area being the culprit. 

Unfortunately its really hard to search github for specific issues IMHO so I can't back up that claim with actual issue links, except this really old one : #6581

Bergerac56, 2017-08-01:
What I just did (and had to stop as I broke the glass bed !. The last time, Z axe went - first time this way- reverse. The head was too close from it. I did not expect such a move and could not halt the printer fast enough...)

Tests done by using G commands only

Files used (see zip: full directory. Exact files used...)

- Start the printer / G28 / G29 L1 / G29 A / G1 X0 Y0 -> G28 homes in the center of the bed. By doing this, I activate UBL and move the head in 0,0 (corner) NO PROBLEM
- G1 X291 Y0 -> All tests below 291 went well. So lets start with X=291. Here the head moves to the choosen point: NO PROBLEM. I can move (G1 X commands with Y=0) all along the X axe (Max X=310) without triggering the Z problem
- G1 X291 Y106 -> Head moves along the Y axe till middle of the Y size: PROBLEM. Z moves without stopping.
- A lot of tests done with other combinaisons of X and Y (using suspected values): At the moment X>291 and Y>105 is reached, it triggers the problem.

Conclusion: 
- With UBL activated, when X >290 and Y>105, Z moves erratically.
- It does not matter from where the head comes: when it reaches X>290 AND Y>105 Z moves.
- The fact that Z increases or decreases seems to be a mix of values of X and Y. For ex: X=300 and Y=200 -> Z increases / X=291 and Y= 110 -> Z decreases.
- The values X>290 seems not to be in the mesh. (I started to repair the printer. I join the mesh file extrated from the ruins ;) )
- When testing: Always have a finger on the "STOP" button...

Now, let's replace the bed...

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1190654/Marlin.zip)
[UBL L1 file.zip](https://github.com/MarlinFirmware/Marlin/files/1190674/UBL.L1.file.zip)

Roxy-3D, 2017-08-01:
@Bergerac56    Can you post a G29 W   and a G29 T ?   That would help a lot to understand the problem.

FHeilmann, 2017-08-01:
@Roxy-3D  the `G29 T` is in his 2nd zip file.

Bergerac56, 2017-08-01:
And here is the result of G29 W
[G29 W.zip](https://github.com/MarlinFirmware/Marlin/files/1190955/G29.W.zip)

Roxy-3D, 2017-08-01:
@Bergerac56     What I'm thinking is we can put in some debug logic that gives us extra information.   What I would do is take you Marlin.zip file up above and add the debug code.    And if you did the same movement to X=291 and Y=105 we would learn a lot about what is going crazy.

Bergerac56, 2017-08-01:
This is OK for me. I am re-assembling the printer just now (I had a spare bed :) ). I suppose it will be finished within half an hour. I suppose you will send me a new "marlin" directory I can compile and load in the printer?

Roxy-3D, 2017-08-01:
Yes...    I'm debating how to proceed...

Roxy-3D, 2017-08-01:
It will be a little verbose...   But if you can do a G28 and then a G1 X291 Y105 and make it fail...

Cutting and pasting the output will give us a clue what is going on....

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1191201/Marlin.zip)

Bergerac56, 2017-08-01:
@Roxy-3D: I am lost . Did you change something?

With the repared printer, I first loaded MY marlin FW (sent in the zip file I posted and recompiled from the zip file to be sure. Same mesh no changes). The problem is there as already described.

Then I loaded YOUR version of marlin FW (The one you sent me. No changes) I first executed  the move you asked (G28 / G1 X291 Y105). Then, in the same session, I did what normally would fail.  And... I cannot reproduce the problem. It does not fail anymore...

I join the output from pronterface:
[Pronterface 01.zip](https://github.com/MarlinFirmware/Marlin/files/1191892/Pronterface.01.zip)

Roxy-3D, 2017-08-01:
if you do a diff of ubl_motion.cpp that is where all the changes are.   I just added a debug information to all the mesh cell paths...

But none of the logic is different.   It just prints out a lot more information about what is happening.

Bergerac56, 2017-08-01:
Then I am really lost :). I just redid the 2 tests with the 2 versions. Mine fails and yours works.

As it is night here, I will sleep on it. And compare each file tomorrow, unless you have another idea.

Roxy-3D, 2017-08-01:
There may be something else going on...    Mine is really yours but with some extra debug print out.   It is possible if you shut down your computer....  And start again with yours...   And load that into the controller board...   things start behaving again.     

I don't know.   But we  might be fighting some problem in Ardruino.

Bergerac56, 2017-08-01:
I shaked it a bit (see log) 
[pronterface02.txt](https://github.com/MarlinFirmware/Marlin/files/1192014/pronterface02.txt)
Your version is solid... till now. I will try what you suggested. I will restart the computer, recompile and reload. Let's see...

Bergerac56, 2017-08-01:
I restarted the computer, recompiled and reloaded my version of marlin. Still fails. 

Tomorow, I will uninstall arduino and install a fresh copy. I will also erase the EEPROM and redo a full install on the printer (new mesh, etc...)

Bergerac56, 2017-08-02:
@Roxy-3D : Hello Roxy. Based on 
- a fresh install of arduino on another PC and with only u8glib added, 
- A full erase of the EEPROM of the printer and a new mesh

I have now a quite similar reaction of your and my marlin directory. Globally: mine does not fail each time anymore (I have to shake it a bit but it happens quickly) and yours fails also (same as mine)

Here are the logs. The number indicated the sequence. The "NOERR" indicates that, this time, I did not insist enough to make it fail.
[Pronterface.zip](https://github.com/MarlinFirmware/Marlin/files/1193878/Pronterface.zip)

I see also that, when the printer initializes, the UBL location 9 in loaded. Is there a reason for that?. (I stored a copy of the mesh there to see if there was a difference. No difference)

When X and Y are in suspected values, time to time, Z starts to move  (+- 1 to 5 mm) No reporting in the log is generated and Z stops (no need of urgent kill). I indicated in one of the logs where it happened.

Arduino is certainly for something in the problem as the behavior is different with a fresh install (Now both versions work the same way). But there seems to be something more... (still no fail with the old way to define the bed size)

Roxy-3D, 2017-08-02:
THANK YOU @Bergerac56  !!!    I don't know what is going wrong yet...   But we are getting very close to the problem!    One thing I'm a little surprised by is NAN is not happening as a Z correction result.  I'll have another version for you to try in a bit.    And with the debug information we get from the next version we can probably put this bug to bed and be done with it!

Bergerac56, 2017-08-02:
Ready to go :)

Roxy-3D, 2017-08-02:
If you can make this one fail (doing the same things you did to the previous one)...   We will know exactly where the problem is.   

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1194342/Marlin.zip)

Bergerac56, 2017-08-02:
Let's try...

Bergerac56, 2017-08-02:
Bingo. It fails happily ;) (The finger on the red button...)

3 trials, 3 fails...
[ROXY V2.zip](https://github.com/MarlinFirmware/Marlin/files/1194450/ROXY.V2.zip)

Roxy-3D, 2017-08-02:
OK!   I think this will fix the problem.    I left the debug code in so that might be annoying but you shouldn't have to have your finger on the reset switch.

If you do the same tests....   You will see the debug code complain that stuff isn't right.   But the Z-height shouldn't do weird things.    It would be good to see 1 or 2 of those debug logs because I expect we are going to see z0 become a NAN number.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1194578/Marlin.zip)

If this does what is expected...  I'll strip out the debug code and get the changes uploaded to bugfix-v2.0.0

Bergerac56, 2017-08-02:
@Roxy-3D 

I know you will hate me...

I was able to make it fail. Less easier than before but still... First attempt went well.

[ROXY V3.zip](https://github.com/MarlinFirmware/Marlin/files/1194762/ROXY.V3.zip)

Roxy-3D, 2017-08-02:
But did the Z axis start going up high or try to crash into the bed?    

That is the important question.

Bergerac56, 2017-08-02:
Yes of course. When you see "Error:KILL caused by KILL button/pin" it is because I had to act quickly (no spare bed anymore ;) )

Roxy-3D, 2017-08-02:
OK...   One last try...  I think this one will do the right thing.    The debug information is still in there so that will give you some clues that you tripped the condition where there used to be a problem.

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1194812/Marlin.zip)

Bergerac56, 2017-08-02:
[ROXY V4.zip](https://github.com/MarlinFirmware/Marlin/files/1194902/ROXY.V4.zip)
Voila. Tests done.

I did around 6 trials (Copied 2 in the zip file) no fails (You are a chief ;) )

One observation (You will see that I played a lot around the boundaries of the previous suspected values)
When the Y axe comes across the value of 190 or the X axe across the value of 290, there is a little jump (closer to the nozzle) of Z. (around 1/4 of motor turn. Did not calculate how much it is). After those limits, no correction of Z at all. This could make sense as those "points" are out of the mesh. I believed that UBL was extrapolating those points. Here it is like Z offers you a last "jump" and stay still after (till you come back in the mesh fiels)

I will do a last test (printing a cube encompassing the "suspected values") and see what happens

Roxy-3D, 2017-08-02:
> After those limits, no correction of Z at all. This could make sense as those "points" are out of the mesh.

There is some controversy on this point.   Some people feel that when outside the mesh, the closest mesh point's value should be used.   Other people feel the correction is undefined and no correction is the appropriate thing to do.   Still others say...  Let's interpolate and use the appropriate amount determined from the two closest mesh points.  Still others say, lets extrapolate the value from the line determined by the outside mesh point and the next inner mesh point.

An argument can be made for each of those strategies.   But really...   If you are moving the nozzle off the mesh, maybe you should define the mesh area to be bigger????    Computationally, that is the simplest.

I'll strip the debug code out and get these corrections folded into bugfix-v2.0.0

Bergerac56, 2017-08-02:
I agree with you. In fact I never printed a large enoug element to go out of the mesh. It is by chance that I found the problem you just solved. (I was preparing a piece for

As a side test, I tried to print the famous testcube, out of the mesh. With the last code you sent me, the printer craches at activation of UBL . I will do some more tests on that.

Bergerac56, 2017-08-02:
Oh, by the way: As said above, I tried bugfix 2.0.0 It compiles fine but the graphic screen does not react. I can pilot the printer with pronterface but the LCD is "blank". Did I miss something?

benlye, 2017-08-02:
Can the fix in #7413 be applied to bugfix-1.1.x as well?  I ran into this issue (well, I think it was this issue) earlier tonight when tinkering around with UBL - I asked my nozzle to move just outside the bed area and the Z axis started to rise to the top.

I'm not quite ready for bugfix-2.0.x yet.  Is it as simple as dropping the two changed files into a bugfix-1.1.x branch?

Roxy-3D, 2017-08-02:
No...  bugfix 2.0.0 has problems.   It is a stepping stone to getting 32-bit support into the bugfix branch.    And the changes I just made a Pull Request to bugfix 2.0.0 don't seem to work with Travis.   

If you want.... You can go into ubl_motion.cpp and comment out the debug code so it doesn't cause so much chit chat.   

> Can the fix in #7413 be applied to bugfix-1.1.x as well?

Yes...    It should do the same thing there.

Bergerac56, 2017-08-02:
Ok I stay with bugfix 1.1.x and apply the changes there. 

Thank you for those corrections. Nice job

I suppose I can close the issue now.

Bergerac56, 2017-08-02:
@Roxy-3D 
I made the changes in bugfix 1.1.x (last version). It works as expected. Thanks again

Roxy-3D, 2017-08-02:
I'm trying to merge the changes into bugfix-v1.1.x right now...    So you shouldn't have to do the manual edits next time you grab the current code base.

Bergerac56, 2017-08-05:
@Roxy-3D Hello Roxy. I reopen the issue to not polute with a new one. I think what I will explain is related.

Doing some tests (again ) I saw that the ubl compensation was not happening between X=0 and the second ubl point on the X axe( The issue is not affection the Y axe it seems) . The first point seems to be skipped. It appears as a "." (dot) on the G29 T0 and as "-----" on the screen (mesh edit). I did not notice that when we made the tests about your corrections as I was mainly using a mesh created with the faulty version. (so the tests were reproductible)

Let's take my case as an example (you have my config files) The print bed is 310 x 210. When issuing a G1 X0 Y0, the nozzle is well in the corner of the print bed. (Looking to the bed: 0,0  is in the upper right corner. This means that 310,210 is near me at left side). The X offset of the probe is 14mm. As there are clips to hold the glass bed, I have a  #define UBL_MESH_INSET 20 and as I want to have 9 points on a row #define GRID_MAX_POINTS_X 9 .

With the last bugfix of today (and already all since the correction you did I think) the first point getting a value is X=053.75. Normally I would expect a first point at X=020.00. This is not the case anymore. When I reload a old firmware (pre-correction) I have well a value in X=020.00

As a consequence, there is an entire zone starting in 0,0 till 53.75, 210 (in my example) without Z correction. 

I know your point of view on zones out of the mesh :) but here ...

If this is not my mistake (perhaps I miss something) I suspect an issue with an incremental variable of which the behavior would have changed after the correction (starting from 1 in place of 0 for ex.)

Roxy-3D, 2017-08-05:
Can you post a G29 T and G29 W of your current setup.   The "." (dot) is where?

Bergerac56, 2017-08-05:
@Roxy-3D . Sorry to react late. I had to reopen the printer again. I noticed that I was unable to load settings from the eeprom. In fact I could not access the eeprom even with other programs than marlin. I suppose I killed it (old arduino mega. too many writes).  What I reported was perhaps some values not correctly stored before definitive dead. I changed the arduino. With the new one, all points are there. No dots and no "-----" on the first raw on LCD (mesh edit). Sorry for that.

And for the last point on the position of the nozzle and the probe. I believe I understood. The position of the probe (built of the mesh) is swapped with the position of the nozzle (printing with UBL). Which makes sense as the only vision UBL has from the bed is through the probe. Moreover I made a small approximation error of the distance between the nozzle and the center of the probe, This is why the "swap" was not perfect and triggered my question.

Once again sorry for the first point.

Roxy-3D, 2017-08-05:
OK!   Sorry your board died!!!

Bergerac56, 2017-08-05:
Chineeze one ;)

I re-close the issue

Bergerac56, 2017-08-06:
@Roxy-3D Hello Roxy. I come back on yesterday. The arduino died for sure, but it was not the cause of the issue with the mesh. It was a coincidence. I get the issue again today with a new arduino... (When I tested yesterday, I got once a normally populated mesh, but today, not anymore)

Please look above for the description of the issue. (Some raws and/or columns are not populated-> "dots". )

I used mainly the two last Bugfix versions (of yesterday and today =lastBF and the included marlin directory). I get always the same results: some points are not populated. (See MESH ERR 01-07.txt). The starting point of the tests was a arduino with a fully erased eeprom (see MESH empty eeprom.txt)

I tried also something else: on a fresh mesh (with "dots") I applied a G29 P3 1. Just to see what would happen. And, surprize, no "1" replacing the dots but the missing raw and column appeared (no dots). (Just a bit puzzled to see 2 nearly identical first raws (not colums) except the last value). See MESH lastBF ERR08.txt. If I save this mesh, it stays in EEPROM (as expected...) See MESH lastBF ERR09.txt location 2

[Mesh issue 01.zip](https://github.com/MarlinFirmware/Marlin/files/1203092/Mesh.issue.01.zip)

I do certainly something wrong . But what?

NB: do not be surprized by one value >1 in the meshs. It comes from a clip maintaining the glass bed. I did not correct it.

Roxy-3D, 2017-08-06:
You have
```cpp
#define X_PROBE_OFFSET_FROM_EXTRUDER 18  // X offset: -left  +right  [of the nozzle] Jacbot = 14
#define Y_PROBE_OFFSET_FROM_EXTRUDER -2  // Y offset: -front +behind [the nozzle]
#define Z_PROBE_OFFSET_FROM_EXTRUDER -0.64  // Z offset: -below +above  [the nozzle] Pour JacBot: -Haut +Bas 

  #define UBL_MESH_INSET 1        // Mesh inset margin on print area   JacBot
```
With these settings...  The probe can't reach (1,1) to probe.   So that line of dots shows up in both directions.

Bergerac56, 2017-08-06:
@Roxy-3D . Hello Roxy,

You are right. I did not notice that I copied back the "default parameter" for the UBL_MESH_INSET.

Thank you for having pointed me to that. It was evident and I did not see it.

closed 2017-08-07 by Bergerac56