MarlinFirmware/Marlin #7377: ### Bug Report

Proto-Plastik, 2017-07-29:
### Bug Report

- Description: 

I have dual Z-axes connected to MKS Base 1.5 that I want to have discreet drivers for. I have wired up the second Z axis to E1. I have enabled all the options in Marlin. This is able to drive the second Z axis with the E1 driver, but the steps per unit is way out of whack. It seems as though it's 10x the speed of the Z axis running off the Z driver.

Where do I set the steps for this second axis? I have tried doing so by defining a second E within Marlin, but the firmware recognizes that I've defined a second extruder and tries to use the next in line that's not in use (E2). It does not seem like the firmware is honoring the steps per unit of the Z driver.

Please do not suggest that I run both z-axes from a single driver. I have been doing this (both parallel and serial) and the performance is not what I need.

Marlin has the capability of setting this and I want to take advantage of it.

- Expected behaviour: 

Both Z axes move at the same speed

- Actual behaviour: 
Z axis from E1 moves at much higher speed

configuration.h and configuration_adv.h below:

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1185109/Configuration.zip)

Proto-Plastik, 2017-07-30:
I am assuming Marlin will use the DEFAULT_AXIS_STEPS_PER_UNIT that is set for the Z-Axis, is that correct? Or, is there a separate location for this setting? Would it be possible that I need to set  the steps for E1 to do this? i.e. {80, 80 , 1518 , 90 , **1518** }

Any help would be greatly appreciated.

gavatron3000, 2017-07-31:
i assume youve enabled //#define Z_DUAL_STEPPER_DRIVERS in configuration_adv.h right?

Proto-Plastik, 2017-07-31:
Hi Gav,

Yes, I've enabled all that stuff. As mentioned in the OP, both Z steppers are working (which would not be the case if Z_DUAL_STEPPER_DRIVERS was not enabled). It's just that the one running off E1 is spinning about 5x (not 10x like I mentioned) faster than the stepper running off of Z. It's possible there's something wrong with the board, but I'm not sure how to check E1. E0 is working fine.

I did not edit pins.h, just pins_ramps.h. 

Still, I don't think that could be it as the motor is definitely turning. It just seems like it's not reading the steps from the DEFAULT_AXIS_STEPS_PER_UNIT setting. Tried the setting I mentioned and it didn't change anything.

For now, I wired up the steppers in series off the Z connector. This is not ideal, but I need to wrap up the machine I'm working on. 

I'm building another straight away so I will be able to test with a different board soon.

gavatron3000, 2017-08-01:
Your microstepping set correctly on both stepper drivers?

fiveangle, 2017-08-01:
your stepper drivers have to have the same microstepping config.  swap cables of your z steppers and see if the problem follows the swap

apphead, 2017-09-15:
I have the exact same issue like @Proto-Plastik with Marlin 1.1.5 . Tried his suggestion to add a value to DEFAULT_AXIS_STEPS_PER_UNIT   {80.4,80.4,404,90,**404**} resulting in compiler error.
Version 1.1.0-RC5 works fine.

Proto-Plastik, 2017-09-15:
Sorry it's been a while since posting here.
It's possible the microstepping is off. However, on the MKS Base 1.5 board, you can't set microstepping so if it's off then they messed it up somehow on the board. It's fine on E0, so I would be surprised if it was wrong on E1. I have a dozen of these boards and have not had any issues whatsoever with them but I'm not ruling it out.

I decided to wire them up in series. I must have done this wrong in the past because they work just fine in that configuration now. Tried parallel first and the low current caused the steppers to skip. Not good.

That said, if 1.1.0-RC5 is working, perhaps I'll give that a try on my next build. 

Thanks for all the replies!

fiveangle, 2017-09-15:
@Proto-Plastik - you need to close - thx

Proto-Plastik, 2017-09-15:
ahhh... sorry about that. Wasn't sure of the protocol :)

Thanks.

closed 2017-09-15 by Proto-Plastik

apphead, 2017-09-17:
Ended up wiring my z-motors in series as well. But in my opinion, the problem with the firmware is not resolved....1.1.0-RC5 works, so it must be concluded that there is something off with the firmware as it is the only factor that changed....