MarlinFirmware/Marlin #7382: [FR] UBL Allow manual edit of unreachable mesh points

Qgel, 2017-07-30:
If I understand UBL correctly, the Z offset at a specific position is determined by interpolation between the mesh points closest to that position. Thus, unreachable points in the mesh (especially on delta printers) have an influence on the calculated Z offset. However, these mesh points cannot be manually adjusted with G29 P4 after they have been generated with G29 P3. This can make fixing leveling issues at the border of the mesh hard.

A command to adjust unreachable mesh points (without moving the nozzle of course) would help in these cases.

Roxy-3D, 2017-07-30:
On Delta's, you should have at least one mesh point surrounding the Printable Radius filled in.    The G29 P3 is modified to handle the Delta case better.   

I agree we need to allow the user to easily edit mesh points that are outside of the area the nozzle can reach.   But in the mean time, you can use M421 to set any of those points that need a different value.

oldmcg, 2017-07-30:
Yes, an interesting and valid point.

We have plans to probe (auto and manual) at the very edges of the reachable radius nearest the grid points that lie outside the radius in order to more accurately extrapolate those unreachable grid points, and manual edit of those unreachable points through P4 would be good for completeness.

GibaHC, 2017-08-02:
I am using `M421` to edit the mesh. As my probing was far from good, pasted the `G29 T1` to excel, and created a VBA module to write the G-Code for the mesh into a text file. Work's good that way. Really faster to edit several points.
I painted the cells to different colors, so i can easily see what type of point i am editing.

Roxy-3D, 2017-08-02:
We have a number of ideas how to edit mesh points off the Printable Radius.   Any one of them would work.   As soon as somebody has some time, somebody will get one implemented.   And if you can write C code...   Somebody will coach you to get it done.   It really isn't that much code and it isn't very difficult code to write.    But it takes time and the person has to be in the mood to do it.

GibaHC, 2017-08-03:
@Qgel , if you have edited one point on the edge of the table, and the previous interpolated out of the bed point is a problem, you can invalidate only that point with something like `G29 X 60.0 Y -60.0 I` and then call `G29 P2` again. If the edges points are good, probably the outer points are going to be calculated correctly.

closed 2018-02-09 by thinkyhead