MarlinFirmware/Marlin #7383: where did "change filament" go?

thomllama, 2017-07-30:
### Bug Report?
- Description:  Change Filament no longer in LCD menu
- Expected behaviour:  be able to change filament using LCD/click wheel
- Actual behaviour: none
- Steps to reproduce:
  - Do this
  - Do that

Recently updated Marlin and just went to use the "change filament" option in the LCD menu and it is gone. :(     Or is it now something I have to turn on?  I got to say the newest Config h file is a tad confusing in it's order/layout.  had a hard time matching settings from my old version to new version.

Hypertoken, 2017-07-31:
If you've enabled "ADVANCED_PAUSE_FEATURE" in the firmware, then you must also have pre-heated your hot end before you can do "Change Filament"

thomllama, 2017-07-31:
can't find any setting anywhere with that title.. did a complete search after looking using textWrangler and nothing comes up.  looked both ConfigH and Adv. config H

Roxy-3D, 2017-07-31:
Its at the bottom of the 'Tune' menu.   It will be the very last item in that list.
```cpp
  /**
   *
   * "Tune" submenu
   *
   */
  void lcd_tune_menu() {
    START_MENU();
...
    //
    // Change filament
    //
    #if ENABLED(ADVANCED_PAUSE_FEATURE)
      if (!thermalManager.tooColdToExtrude(active_extruder))
        MENU_ITEM(function, MSG_FILAMENTCHANGE, lcd_enqueue_filament_change);
    #endif

    END_MENU();
  }
```
But it will only be there if ADVANCED_PAUSE_FEATURE is turned on and the active extruder is not too cold.

Hypertoken, 2017-08-01:
@thomllama If you aren't up to date on firmware, the "ADVANCED_PAUSE_FEATURE" has recently been renamed from  "FILAMENT_CHANGE_FEATURE"

closed 2018-02-09 by thinkyhead