MarlinFirmware/Marlin #7385: ### Bug in Case Light in 2.0x

DaveAhrendt, 2017-07-31:
With 2.0.x on a FolgerTech i3 2020 the behavior of the case light menu item is broken.  You can only use the menu selection to turn on/off the case light if the brightness is set to 255.  Any other value and the on/off is ignored.
The function worked properly in the earlier 1.1.x releases.
This is on a standard Mega 2560 and Ramps 1.4.

Hywelmartin, 2017-12-22:
Is also present in 1.1.6 
M355 S0/S1 works only if brightness is set at P255

Mega clone(anycubic delta)

thinkyhead, 2017-12-25:
How about `bugfix-1.1.x` @Hywelmartin ? Version 1.1.7 is also out now.

maukcc, 2018-01-23:
We are having a separate (compile) issue with an Arduino Due : 

> sketch/src/feature/caselight.cpp: In function 'void update_case_light()':
> sketch/src/feature/caselight.cpp:73:44: error: 'USEABLE_HARDWARE_PWM' was not declared in this scope
>      if (USEABLE_HARDWARE_PWM(CASE_LIGHT_PIN))

Bob-the-Kuhn, 2018-01-23:
bugfix-2.0.x ?
Arduino 1.8.5 ?

Bob-the-Kuhn, 2018-01-24:
I looked a little harder at the error message and realized you were using bugfix-2.0.x and Arduino.

I Just successfully compiled bugfix-2.0.x with Arduino 1.9.0 beta.

Please download a new copy of bugfix-2.0.x and see if that cures the problem.

maukcc, 2018-01-24:
It is our custom board that is the culprit.
Any special things that need doing for a new board?

maukcc, 2018-01-24:
Here are our changed files:
[Marlin-bugfix-2.0.x-changed.zip](https://github.com/MarlinFirmware/Marlin/files/1659105/Marlin-bugfix-2.0.x-changed.zip)

Bob-the-Kuhn, 2018-01-24:
Your changes look good.

I dropped your changed files into yesterday's bugfix-2.0.x and it compiled without errors with both Arduino 1.9.0 beta and PlatformIO.

If you're using the latest bugfix-2.0.x and still having compile issues then ZIP up EVERYTHING plus the complete Arduino log and post it.

----

When bringing up a new board I like to use the M43 command.  You need to enable **PINS_DEBUGGING** in **Configuration_adv.h**.
 ```
  /**
   * M43: Pin debug - report pin state, watch pins, toggle pins and servo probe test/report
   *
   *  M43         - report name and state of pin(s)
   *                  P<pin>  Pin to read or watch. If omitted, reads all pins.
   *                  I       Flag to ignore Marlin's pin protection.
   *
   *  M43 W       - Watch pins -reporting changes- until reset, click, or M108.
   *                  P<pin>  Pin to read or watch. If omitted, read/watch all pins.
   *                  I       Flag to ignore Marlin's pin protection.
   *
   *  M43 E<bool> - Enable / disable background endstop monitoring
   *                  - Machine continues to operate
   *                  - Reports changes to endstops
   *                  - Toggles LED_PIN when an endstop changes
   *                  - Cannot reliably catch the 5mS pulse from BLTouch type probes
   *
   *  M43 T       - Toggle pin(s) and report which pin is being toggled
   *                  S<pin>  - Start Pin number.   If not given, will default to 0
   *                  L<pin>  - End Pin number.   If not given, will default to last pin defined for this board
   *                  I<bool> - Flag to ignore Marlin's pin protection.   Use with caution!!!!
   *                  R       - Repeat pulses on each pin this number of times before continueing to next pin
   *                  W       - Wait time (in miliseconds) between pulses.  If not given will default to 500
   *
   *  M43 S       - Servo probe test
   *                  P<index> - Probe index (optional - defaults to 0
   */

```

I like to use the **M43 I** command to see the current state of the pins and the logical signals attached to them.

I also use the "M43 W I" command when I don't know what logical pin is attached to a physical pin.  I'll issue the command and then use a 1K resistor to ground and see what is reported.  Sometimes it takes a 1K to +3.3V/+5V to see something.  Sometimes you see nothing.  In that case you'll need to use the "M43 T" command and an LED.

I also like to remove the protected pin check in M42.  That way you can attach LEDs to pins and then manually toggle them to be sure you have the right pin.

----

Curiosity question - can you provide a brief description of your board?

Bob-the-Kuhn, 2018-01-24:
I just realized that CASE_LIGHT had NOT been enabled in your files.

I'm now seeing the USEABLE_HARDWARE_PWM error.  I'll see what is going on.

Bob-the-Kuhn, 2018-01-25:
Add the following to fastio_Due.h

// Due has 12 PWMs assigned to logical pins 2-13.
// 6, 7, 8 & 9 come from the PWM controller. The others come from the timers.
#define USEABLE_HARDWARE_PWM(p) ((2 >= p) && (p <= 13))

Bob-the-Kuhn, 2018-01-25:
PR #9345 fixes this issue.

maukcc, 2018-01-25:
OK, compiles good now and works as well :)
The thing I am missing from the LCD is that it just does on/of, no brightness. minor detail.

Here is a pdf of our new CNControls V14 board: 
[CartesioCNControls14.pdf](https://github.com/MarlinFirmware/Marlin/files/1663324/CartesioCNControls14.pdf)
It is 24V based board, starting from the top going clockwise, has 4 tools in/outputs, XYZ motor outputs
(no drivers onboard) and enstop inputs, 4x filament encoder/runout, display(nextion) output, raspi serial and power output, 24V input, chamber heater output(100mA), red and green LED output(100mA), case light(100mA), watercooler output(1A), fume extraction output(1A)

more info on www.cartesio.nl

maukcc, 2018-01-25:
Does brightness work at all?
Or are we better of using M42 Pxx Sxx

Bob-the-Kuhn, 2018-01-26:
I had only one error in the line of code.

It should read:
#define USEABLE_HARDWARE_PWM(p) ((2 <= p) && (p <= 13))

I'll update Marlin within an hour.

maukcc, 2018-01-26:
That worked in getting a lcd menu item with brightness, unfortunately the pin can not be set anymore through M355.
M42 P13 S9 still works

Bob-the-Kuhn, 2018-01-27:
Isn't it fun being the first one to try something?

In **fastio_Due.h** replace the  **_SET_OUTPUT** macro with this one.

I verified it works on mu Due board.

```
/// set pin as output
#define _SET_OUTPUT(IO) do{ pmc_enable_periph_clk(g_APinDescription[IO].ulPeripheralId); \
                            PIO_Configure(g_APinDescription[IO].pPort, _READ(IO) ? PIO_OUTPUT_1 : PIO_OUTPUT_0, \
                                          g_APinDescription[IO].ulPin, g_APinDescription[IO].ulPinConfiguration); \
                            g_pinStatus[IO] = (g_pinStatus[IO] & 0xF0) | PIN_STATUS_DIGITAL_OUTPUT;\               
                        }while(0)
```

Bob-the-Kuhn, 2018-01-27:
This change is being implemented via PR #9364

maukcc, 2018-01-29:
That did the trick ! perfect.
1 small thing we would change is: in the lcd menu there is 1st the brightness, then the on/off.
We would do on/off 1st, then brightness.
Or, if the light is off and you change brightness it actually puts it on and changes the brightness.

Bob-the-Kuhn, 2018-01-30:
The case_light code has been changed many times.  Too many cooks so I leave it alone.

maukcc, 2018-02-01:
I hear ya!
Is this issue done for @DaveAhrendt as well?

closed 2018-02-14 by thinkyhead