MarlinFirmware/Marlin #7388: How to set arithmetic operation with marlin

batata004, 2017-07-31:
Hi,

I am using this line of code:

`#define USER_GCODE_1 "M140 S" STRINGIFY(PREHEAT_2_TEMP_BED/2) "\nG28\nG29"`

You see that I am trying to divide PREHEAT_2_TEMP_BED per 2 and use STRINGIFY? The problem is that diving by 2 is not working, the compiler accepts it, compiles but the value is never divided by 2. How can I achieve this?

It's not a bug, it's a usage case.

Roxy-3D, 2017-07-31:
> STRINGIFY(PREHEAT_2_TEMP_BED/2)

I don't know the answer.    I wonder if you put paren's around it? 

```cpp
#define USER_GCODE_1 "M140 S" STRINGIFY( (PREHEAT_2_TEMP_BED/2) ) "\nG28\nG29"
```

That might force the evaluation of what is getting passed to STRINGIFY()  ????

bobc, 2017-07-31:
STRINGIFY converts a text sequence like Abcd to a string by adding quotes: "Abcd", it cannot do arithmetic. It doesn't even know that the string is a number.

It's not impossible to do arithmetic in the preprocessor, but requires a lot of macro code. I think the Boost preprocessor can do it, but we do not use Boost.

Roxy-3D, 2017-07-31:
Thanks for the explanation!

I guess that means batata004 should just hard code the desired numbers and not be too fancy for his USER_GCODE numbers....   Oh well....

batata004, 2017-07-31:
@Roxy-3D I tried using your suggestion (double parenthesis) but it was interpreted as 0 by the compiler.

@bobc explanation was clear! I just wonder why the code below does not work, why the preprocessor does not make arithmetic since it's run in a computer processor which is waaay more powerful than arduino. Why nobody at arduino dev team didnt think about allowing arithmetic operations!

`#define PREHEAT_2_TEMP_BED_DIV_2 (PREHEAT_2_TEMP_BED/2)

`

bobc, 2017-07-31:
The C preprocessor was invented in about 1972....it's a lot simpler than
you think, it's main purpose was to allow #include files.
https://softwareengineering.stackexchange.com/questions/309389/what-is-the-origin-of-the-c-preprocessor
The limitations of the C preprocessor are nothing to do with Arduino.

Expressions in User Gcode could be handled at run-time, at some cost in
space and CPU time. I suppose you could also write a new preprocessor to
handle expressions.

batata004, 2017-07-31:
Thanks, I had a misunderstanding of it! ;)

closed 2017-07-31 by batata004