MarlinFirmware/Marlin #7389: Extruder not working

Dracen-Knights, 2017-08-01:
ok so I'm new to all this and not 100% sure where to find the info I'm looking for.  so anyway I have an MKSgen-L v1.0 with A4988 stepper drivers. I have a firmware file that is known to work with the exact parts I have but I can't get the extruder to feed to calibrate it. I have all the other axis calibrated and working great...  Anyway... anything I should look for in the code to make sure it's done correctly? I have check the wires and steppers them-self on other drivers and they work then swap drivers around and while for example driver X will work good in it's location nothing when I move that driver to the extruder drivers spot. 

Any place to find more info on the coding of this? I'm getting the hang of it quickly just looking for more info. 

also looking at doing a custom boot screen because I can so any place to find info on doing that? kind of a walk through?

Dracen-Knights, 2017-08-01:
BTW my screen is full color so do I have to stay B&W for the custom boot screen? kind of seams a waste as is the factory splash is green

KevMags, 2017-08-01:
Marlin defaults to "no cold extrusion".  The hotend needs to be hot before the extruder motor will turn.

Dracen-Knights, 2017-08-01:
I'm pretty sure I heated the hotend once but I'm trying that now. to see it maybe it wasn't hot enough. I have never used my computer on the printer other than to setup firmware

Dracen-Knights, 2017-08-01:
ok guess I didn't let it get hot enough the first time I did it... This time at about 200 it started to feed...

Kaibob2, 2017-08-02:
Please close here if your problem is fixed.

Dracen-Knights, 2017-08-02:
part of the issue is but I asked other questions in the same post. The extruder driver item is sorted.

Heres the second part again...   also looking at doing a custom boot screen because I can so any place to find info on doing that? kind of a walk through? My tft is full color and the splash screen thats on it is green and white so why can't I do color?

Kaibob2, 2017-08-02:
I suspect full color isn't supported until now. Maybe this will guide you a little by creating a custom boot screen [Link](http://community.robo3d.com/index.php?threads/custom-boot-screen-for-marlin-and-full-graphics-lcd.17221/)

Dracen-Knights, 2017-08-02:
thanks for the links and I have seen both but both only talk about black and white...  guess I'll have to wait until someone figures out the color issue. and frankly those  how-to's are confusing so guess I'll leave it thanks for the links.

closed 2017-08-02 by Dracen-Knights