MarlinFirmware/Marlin #7392: ReprapDiscount Smart Controller - No Screen - Dark

zurpv1, 2017-08-01:
Hi,
I am trying to hookup my ReprapDiscount Smart Controller - rotary encoder and a 20 Character x 4 Line LCD display. - to my MKS Gen 1.4 board. 

The LCD panel is not coming on at all. I did notice that the encoder is sending signals to the board. I have tried latest versions of Marlin and Skynet3d. Any help would be great. 

Thanks

------------------------------------------------------------
### Bug Report
- Description: --- ReprapDiscount Smart Controller on MKS Gen 1.4 latest v of Marlin.
- Expected behaviour: --- LCD comes on and works
- Actual behaviour: --- Blank LCD Screen, No back light, seems to have no power
- Steps to reproduce:
  - Do this - Plug in LCD to MKS board. Flash with Marlin.

[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1190549/configuration.zip)

zurpv1, 2017-08-01:
This video shows exactly what is happening.

fiveangle, 2017-08-01:
many clone lcds/boards have the connector keys backwards, so confirm that first using something like a multimeter or simple continuity tester

zurpv1, 2017-08-01:
Thank you. I will check it out when I get home.

closed 2017-08-01 by zurpv1