MarlinFirmware/Marlin #7393: Latest firmware location - which branch?

playlet, 2017-08-01:
Hi guys!

This is not a bug report or anything like that. I'm just wondering what is the location of the latest marlin firmware here in the github tree?

I see there are now 2.0.x and bugfix 2.0.x branches but bugfix 1.1.x was edited last so I'm not sure what is the "latest" branch I can use to download and test it?

Thanks,

Ivan

mperdue, 2017-08-01:
My understanding is that 2.0.x is very rough at this point. I'm watching to see when switching to it and going with 32 bit might be a workable idea.

fiveangle, 2017-08-01:
latest 1.x (latest release) or bugfix1.1.x (latest-latest)

All 2.x stuff is work-in-progress to incorporate 1.x + 32-bit targets.  Most Arduino targets don't even compile on it at this point.

-=dave

playlet, 2017-08-01:
All right, thanks guys :)

closed 2017-08-01 by playlet

Roxy-3D, 2017-08-16:
What fiveangle said...   Except this is my perspective:   bugfix-2.0.0 is for 32-bit work.   bugfix-1.x.x is for AVR work.    At least right now...   My guess is soon there will only be one bugfix branch again.