MarlinFirmware/Marlin #7396: Compile error with "PROBE_MANUALLY" and "LCD_BED_LEVELLING"

JimBrown, 2017-08-01:
When I compile with `PROBE_MANUALLY` and `LCD_BED_LEVELLING` I get an error:
```
#error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY."
```
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1191589/Configuration.zip)

Thanks.

fiveangle, 2017-08-01:
You have no leveling mechanism enabled:

```
//#define AUTO_BED_LEVELING_3POINT
//#define AUTO_BED_LEVELING_LINEAR
//#define AUTO_BED_LEVELING_BILINEAR
//#define AUTO_BED_LEVELING_UBL
//#define MESH_BED_LEVELING
```

JimBrown, 2017-08-01:
Error says "Mesh bed leveling" OR "Probe Manually". I have "Probe Manually" enabled.

fiveangle, 2017-08-01:
What do you expect to do with LCD_BED_LEVELLING then ?  There is no facility in the FW to adjust the levelling without a manual probe method and levelling scheme.  You must adjust the levelling mechanically external of the FW.

JimBrown, 2017-08-01:
```
/**
 * The "Manual Probe" provides a means to do "Auto" Bed Leveling without a probe.
 * Use G29 repeatedly, adjusting the Z height at each point with movement commands
 * or (with LCD_BED_LEVELING) the LCD controller.
 */
#define PROBE_MANUALLY

```

```
/**
 * Use the LCD controller for bed leveling
 * Requires MESH_BED_LEVELING or PROBE_MANUALLY
 */
#define LCD_BED_LEVELING

#if ENABLED(LCD_BED_LEVELING)
  #define MBL_Z_STEP 0.025    // Step size while manually probing Z axis.
  #define LCD_PROBE_Z_RANGE 4 // Z Range centered on Z_MIN_POS for LCD Z adjustment
#endif
```

fiveangle, 2017-08-01:
> * The "Manual Probe" provides a means to do "Auto" Bed Leveling without a probe.

Yes, emphasis on the "Auto" requirement:

```
/**
 * LCD_BED_LEVELING requirements
 */
#if ENABLED(LCD_BED_LEVELING)
  #if DISABLED(ULTIPANEL)
    #error "LCD_BED_LEVELING requires an LCD controller."
  #elif DISABLED(MESH_BED_LEVELING) && !(----->HAS_ABL<----- && ENABLED(PROBE_MANUALLY))
    #error "LCD_BED_LEVELING requires MESH_BED_LEVELING or PROBE_MANUALLY with auto bed leveling enabled."
  #endif
#endif
```

I've submitted #7398 to resolve the confusion

JimBrown, 2017-08-01:
Got it. Thank you.

closed 2017-08-01 by JimBrown