MarlinFirmware/Marlin #7400: [FR] Add serial port number validation

ellensp, 2017-08-01:
Please add in some serial port validation

the configuration.h currently says
"/**
 * Select which serial port on the board will be used for communication with the host.
 * This allows the connection of wireless adapters (for instance) to non-default port pins.
 * Serial port 0 is always used by the Arduino bootloader regardless of this setting.
 *
 * :[0, 1, 2, 3, 4, 5, 6, 7]
 */
#define SERIAL_PORT 0"

But nothing is checking that the serial port is in that range. 
I found a user setting it to 12 as that was the com port on his computer(sigh)... gave no warnings and lots of compile errors (not surprisingly)

fiveangle, 2017-08-02:
It begs the question if such a person should be compiling their own FW ;)

thinkyhead, 2018-02-09:
Done!

closed 2018-02-09 by thinkyhead