MarlinFirmware/Marlin #7402: G33 std dev is reporting HUGE (BLTouch)

mbabauer, 2017-08-02:
I have an Anycubic Kossel running Marlin bugfix-1.1.x that I have added a BLTouch to in order to get auto bed leveling.  I have the results of `G33 P1` returning desirable highs consistently, but when I run just `G33` it reports a wild "std dev" value:

```
>>> g29
SENDING:G29
>>> g33
SENDING:G33
G33 Auto Calibrate
echo:busy: processing
echo:busy: processing
Checking... AC
.Height:336.40    Ex:-0.10  Ey:+0.00  Ez:-0.11    Radius:97.91
.Tower angle :    Tx:-0.04  Ty:+0.11  Tz:+0.00
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Calibration OK                                    std dev:-2147483.750
.Height:336.40    Ex:-0.10  Ey:+0.00  Ez:-0.11    Radius:97.91
.Tower angle :    Tx:-0.04  Ty:+0.11  Tz:+0.00
Save with M500 and/or copy to Configuration.h
```

I know my towers are square to the best of my ability, and I know by printing the [Advanced Delta Calibration](https://www.thingiverse.com/thing:745523) that the X and Y towers are a bit off at angle (both seem to be ~119.25º).

Running `M48` produces what seems like better results:
```
>>> M48 V1
SENDING:M48 V1
M48 Z-Probe Repeatability Test
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
echo:busy: processing
Finished!
Mean: -0.005002 Min: -0.013 Max: 0.000 Range: 0.013
Standard Deviation: 0.006130
```

Shouldn't the standard deviation on `G33` be a much smaller number?

Here is my current Configuration.h
[Configuration.h.zip](https://github.com/MarlinFirmware/Marlin/files/1192514/Configuration.h.zip)

Roxy-3D, 2017-08-02:
If my BL-Touch always returned a standard deviation of .006 I would be thrilled.   Even with a layer height of .2 mm a standard deviation of .01 will give you very good results.    And then you combine it with Z-BabyStepping and the Double Click to jump to Z-BabyStepping and I assure you...  You won't have problems getting the first layer down with perfect adhesion.

fiveangle, 2017-08-02:
@LVD-AC - can you review this ?  It seems 4bc79ec877e4a4c247f5014d594ca8ca99a974c1 may need some checking.

LVD-AC, 2017-08-02:
There was a problem with eccentric probes returning NAN instead of a value on valid probes (hence the large negative number for the std dev); this error was fixed quit a while ago. But you can fix this by reducing DELTA_CALIBRATION_RADIUS

mbabauer, 2017-08-03:
@LVD-AC Right now it's being calculated using:
`#define DELTA_CALIBRATION_RADIUS ((DELTA_PRINTABLE_RADIUS) * 0.869) // mm`
With DELTA_PRINTABLE_RADIUS set to:
`#define DELTA_PRINTABLE_RADIUS 90.0`
How far would you recommend dropping it to

@Roxy-3D I understood only some of those words :).  Seriously, what is Z-BabyStepping?  I searched config and didn't see any settings containing "baby".

Roxy-3D, 2017-08-03:
> @Roxy-3D I understood only some of those words :). Seriously, what is Z-BabyStepping? I searched config and didn't see any settings containing "baby".

This is something the 'Purists' will cringe at...   If you have a high dollar machine that is perfectly calibrated and can hold tight tolerances and never drift.... And if you use expensive filament from just one vendor that never varies...   You don't need Z-BabyStepping....

But  I deliberately set my nozzle height .1mm or .15mm too high off the bed.  When ever I start a print, I double click the encoder wheel while it is drawing the skirt of the print.   That puts you into Z-BabyStepping and you can micro-adjust the height of the nozzle as it is drawing the skirt to get perfect adhesion.

If you decide you want to get perfect adhesion of your first layer every time...    Just turn this on

https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Configuration_adv.h#L591-L598

fiveangle, 2017-08-03:
TL;DR version : "babystepping" allows to adjust the bed-to-nozzle "zero" coordinate point on the fly in order to fine-tune things, especially the first layer.  Only the graybeards know why it was called "baby stepping" but the odd non-descript name has somehow endured.

LVD-AC, 2017-08-03:
@mbabauer diminish DELTA_CALIBRATION_RADIUS with the probe offset (hypotenuse of XY-values), and set DELTA_PRINTABLE_RADIUS to the maximum you can without the effector running into the belts.

 I would recommend you upload the latest version with this and other bugs for eccentric probes fixed...

@Roxy-3D I just have a G33 C0.03 in my startup code; that does the necessary fine tuning automatically for me; or you could use G33 P1 just to fine tune the height.

LVD-AC, 2017-08-03:
For the latest version 1.1.4 onwards:

- set DELTA_PRINTABLE_RADIUS to the maximum you can without the effector + fans + probe running into the belts
- set DELTA_CALIBRATION_RADIUS to the maximum you can without the probe running off the bed (and limit to DELTA_PRINTABLE_RADIUS * 0.869 to get full use of G33)

For 1.1.3 and below:
- set DELTA_PRINTABLE_RADIUS to the maximum you can without the effector running into the belts
- set DELTA_CALIBRATION_RADIUS to DELTA_PRINTABLE_RADIUS * 0.869 - HYPOT(X_PROBE_OFFSET_FROM_EXTRUDER, Y_PROBE_OFFSET_FROM_EXTRUDER)

Using a larger radius to calibrate increases the calibration precision at the limits of the bed (large objects).

mbabauer, 2017-08-07:
@LVD-AC 
I have `DELTA_PRINTABLE_RADIUS` set to 90 right now, which seems to allow the printhead to come as close as it can without actually hitting the belts.  Should I upgrade to Marlin 1.1.4 and see if that helps?

mbabauer, 2017-08-17:
I upgraded to the latest version (1.1.4), transferred all my settings that matched up (some I guess are no longer available) and now I seem to be getting more normal standard deviation:

```
echo:busy: processing
...
echo:busy: processing
Iteration : 03                                    std dev:0.122
.Height:336.70    Ex:-0.45  Ey:+0.00  Ez:-0.90    Radius:97.89
.Tower angle :    Tx:-0.11  Ty:+1.11  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
Iteration : 04                                    std dev:0.092
.Height:336.70    Ex:-0.54  Ey:+0.00  Ez:-0.96    Radius:98.03
.Tower angle :    Tx:-0.02  Ty:+1.03  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
Iteration : 05                                    std dev:0.095
.Height:336.72    Ex:-0.58  Ey:+0.00  Ez:-1.00    Radius:98.11
.Tower angle :    Tx:+0.08  Ty:+1.05  Tz:+0.00
echo:busy: processing
...
echo:busy: processing
Calibration OK                                    rolling back.
.Height:336.70    Ex:-0.45  Ey:+0.00  Ez:-0.90    Radius:97.89
.Tower angle :    Tx:-0.11  Ty:+1.11  Tz:+0.00
Save with M500 and/or copy to Configuration.h
echo:busy: processing
...
echo:busy: processing
```

One issue I have still is that in the back-left prob (closest to Z tower, slightly -X direction) the head seems to hit prior to the prob hitting.  The one before that the head touches right as the prob triggers as well.  All the other points seem to prob just fine, it's just those two.

LVD-AC, 2017-08-17:
> •set DELTA_CALIBRATION_RADIUS to the maximum you can without the probe running off the bed 

Still to high it seems

LVD-AC, 2017-08-23:
#7524 
> I do have a ticket open, #7402, due to G33 not really working as well (my standard dev was not right...seemed to be the version of the FW I was using). I believe this problem is "fixed" now,

Can be closed then