MarlinFirmware/Marlin #7403: FR: UBL: Help / Documentation needed configuring probe and print area

christianh17, 2017-08-02:
I become desperate configuring UBL, print area and probe area. I tried so many things but either not all the bed is probed or the inductive probe is outside the bed and of course the printhead crashes into the bed. Or I get compile errors because the UBL_PROBE_POINTS cannot be reached.

To make it a little bit easier let us look only into the x-axis, of course the problem is the same for the y-axis but it would not help regarding both.

Marlin Version: bugfix 1.1.x, last downloaded yesterday
Hardware: Prusa compatible 3D Printer, switching nozzle, inductive probe
bed size: 220*220 mm
All distances are measured regarding the nozzle:
Movement limits: X from 0 to 246 mm
Print Limits: X from 28 to 246 mm
Probe-Offset X:  45 mm

So I need help how to configure the following setting:
#define X_MIN_POS (is 0 correct? or do I have to fill 28?)
#define X_MAX_POS (246 must be correct)
#define X_PROBE_OFFSET_FROM_EXTRUDER (45 must be correct- what else should I enter here)
#define UBL_MESH_INSET 1 - I did not change it
#define UBL_PROBE_PT_1_X (is 29 correct for the leftmost point?)
#define UBL_PROBE_PT_2_Y (is 246 correct for the rightmost point?)

I do not understand if i have to fill in the ubl_probe-setting regarding the nozzle or the probe. And: I do not understand where to configure the printing area...

Thanks a lot in advance,
Christian

benlye, 2017-08-02:
I have a similar printer, where the X_MIN_POS is off the bed, and am just starting to bring up UBL.  

It seems to me that the trick is to use the new _BED_SIZE parameters.

My configuration.h has lines like this (just looking at X, values updated to match yours):
```
#define X_BED_SIZE 220
#define X_MIN_POS -28
#define X_MAX_POS 246

#define UBL_MESH_INSET 10
```

Then in configuration_adv.h:
```
  #define UBL_MESH_MIN_X (0 + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_BED_SIZE - (UBL_MESH_INSET))
```

I know @Roxy-3D doesn't agree that using BED_SIZE in this way is the right thing to do, but it works for me (and makes more sense to me then using the MIN/MAX position).

oysteinkrog, 2017-08-02:
@benlye: this is exactly what I do as well, and it works great for me w/UBL.

christianh17, 2017-08-02:
Perfect, that works perfect.
I did not have the time to do the 3point-levelling but all the other things: great!
The idea with a negative x-value at the homing point is interesting, too, That helps!
@Roxy-3D : What is the way you suggest how to do the configuration?

Again: Thank you very much! @benlye 👍 @oysteinkrog 👍 
Regards
Christian

Roxy-3D, 2017-08-02:
> @Roxy-3D : What is the way you suggest how to do the configuration?

If you have it working with the ?_BED_SIZE  defines...  Just leave it.  I don't think those are going away.  My biggest concern is I don't believe BED_SIZE is being used correctly to define the SAFE_HOMING point at the center of the bed.    That is mostly what I'm concerned about.

Kaibob2, 2017-08-03:
I had a related problem with `MESH_BED_LEVELING`
In my setup the X-endstop is at -32. 
[image]
When i run `MESH_BED_LEVELING` the "left" nozzle moves ` #define MESH_INSET 10` to the right, but this is still outside of the printing area. Setting `#define MESH_MIN_X (X_MIN_POS + MESH_INSET)` to `#define MESH_MIN_X (0 + MESH_INSET)` fixed this perfectly.
@Roxy-3D Wouldn't it be reasonable to change
```
#if ENABLED(MESH_BED_LEVELING)
  #define MESH_MIN_X (X_MIN_POS + MESH_INSET)
  #define MESH_MAX_X (X_MAX_POS - (MESH_INSET))
  #define MESH_MIN_Y (Y_MIN_POS + MESH_INSET)
  #define MESH_MAX_Y (Y_MAX_POS - (MESH_INSET))
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
  #define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET))
```

to

```
#if ENABLED(MESH_BED_LEVELING)
  if (X_MIN_POS < 0)
  #define MESH_MIN_X (0 + MESH_INSET)
  else
  #define MESH_MIN_X (X_MIN_POS + MESH_INSET) 
  if (Y_MIN_POS < 0)
  #define MESH_MIN_Y (0 + MESH_INSET)
  else
  #define MESH_MIN_Y (Y_MIN_POS + MESH_INSET)
  #define MESH_MAX_X (X_MAX_POS - (MESH_INSET))
  #define MESH_MAX_Y (Y_MAX_POS - (MESH_INSET))
#elif ENABLED(AUTO_BED_LEVELING_UBL)
  if (X_MIN_POS < 0)
  #define UBL_MESH_MIN_X (0 + UBL_MESH_INSET)
  else
  #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
  if (Y_MIN_POS < 0)
  #define UBL_MESH_MIN_Y (0 + UBL_MESH_INSET)
  else 
  #define UBL_MESH_MIN_Y (Y_MIN_POS + UBL_MESH_INSET)
  #define UBL_MESH_MAX_X (X_MAX_POS - (UBL_MESH_INSET))
  #define UBL_MESH_MAX_Y (Y_MAX_POS - (UBL_MESH_INSET))
```

Roxy-3D, 2017-08-03:
Maybe....     But if we were to do that we would need to use some form of the pre-processor #if
```cpp
#if  (X_MIN_POS) < 0
```
A static assert might work there.    The problem is those 'if' statements are going to try to generate code and not do what you expect.

Kaibob2, 2017-08-03:
I see. Then what about this:
```
// Default mesh area is an area with an inset margin on the print area.
// Below are the macros that are used to define the borders for the mesh area,
// made available here for specialized needs, ie dual extruder setup.
// For setups where the endstop position is < 0 set MIN_POS = 0 here
// Example: X-endstop is at -30 then exchange #define MESH_MIN_X (X_MIN_POS + MESH_INSET) with #define MESH_MIN_X (0 + MESH_INSET)
// The same applies to UBL
```
Something like this would have saved me a lot of time figuring out how to get this working.

Roxy-3D, 2017-08-03:
Part of the problem is "what does inset" mean.   For me it means I can remove part of the bed's print envelope.  What ever the size of the bed is...   Inset lets the user prune back the size.    Usually that is not desirable.   But for people that need it, it is very helpful.

It maybe with the new X&Y BED_SIZE we can eliminate the inset complexity.   In that case, the person would just specify how much of the bed they want to use and where its center is.  

But for right now, we inset is still part of the equation.    Can you take a stab at trying to make it clear how inset is (or can be) used by the system.

Kaibob2, 2017-08-03:
For me it was clear what inset means from the beginning. When i have a printbed which is 200x200 and the inset is set to 10 then the area which gets probed is 180x180.
Probing starts at at X10 Y10 and the most far away point from there is X190 Y190.
This is part of the problem with `#define MESH_MIN_X (X_MIN_POS + MESH_INSET)` 
With this line active probing starts 10mm away from the HOMEPOINT. If the homepoint is at -30 probing starts at -20. Unfortunately there is no bed to probe because the bed starts at 0.
I think inset is quite a good thing, because it might not be the best idea to probe the path of the absolute bed size.
In practice, for me, i can use an effective bed size of X217 and Y200 but i don't want to to probe the bed at exactly X217 but maybe X215.
This all mabe different when using a capacitive probe or equal. I use Manual mesh leveling, as my bed is already extremely even.

fiveangle, 2017-08-04:
Curious: why would one not want to probe "inset" of their print area and not the printable area ?

Kaibob2, 2017-08-04:
I see your point. Maybe it would be meaningful to rename `// The size of the print bed` to `// The size of the printable area`

Regarding inset: I would say it is nice to have but not essential. For example the picture below is from my printer. X0 is there where the left nozzle is. The endstop is at -32. It surely is possible to probe there, but it's easier to probe a few mm more to the left.
Regardless of probing, i can print there.
[image]
If inset woudl be discarded, printing at X0 (X0 is there where the left nozzle is) wouldn't be possible anymore as the printing area is smaller than the actual bed size.

A little confusing subject this is...

Roxy-3D, 2017-08-04:
The reason inset is in UBL is because it was in the original mesh bed leveling system.   If it wasn't for that reason I wouldn't have included it.

But with that said...  I use inset on my gMax printer.   It has a very large bed.  I wanted to use a silicon heat pad with borosilicate glass.    I could not get a either of them (easily) for the full size.   But I could get each of them at 400mm x 400mm.     

So, my printer homes off the bed.     And with a 45mm inset, it does the right thing.

christianh17, 2017-08-04:
In my opinion we have different dimensions we have to think about:

- The physical limits of the printer, so limits of movement. Normally (I only know about certesian printers) one limit per axis is physically set by the endstops, the other limit should be set by software
- print limit: Where on the bed can be printed - it is obvious what is meant
- probing limit: Where on the bed can be probed. In my setting the probe is quite big, it is a rectangle from about 10 by 5 mm. To be sure it works correctly all of the probe should be over the bed during the probing. So here we have the reason why an inset is reasonable
- and of course the offset nozzle -> probe
- In my case there is another dimension, my switching extruder (dondolo style) needs offsets too, but this is not the standard setting.

I personally would prefer configuring all these dimensions individually. The actual possible configuration tries to do a lot of automatic calculations. These are are difficult for all possible settings.
And: I still do not know for sure which setting will be normalized against the probe-offsets and which will be used as it is. So: A clear naming convention for absolute or relative coordinates would be helpful.

What do you think about this idea?

epatel, 2017-08-04:
The reason I added the inset was that on my big bed I have a number of clips that hold down a glas sheet and with the inset I make sure to travel/probe a safe distance from these clips. Also, in the original implementation the math outside the print area was extrapolated from the closest mesh area, i.e. a downward slope continued downward which I think worked fine when printing on my printer. Feel free to remove the insets if you want, seems like the extrapolation has been removed so that makes more sense then too (might be wrong as I am only reading here occasionally).

Kaibob2, 2017-08-07:
I agree to christianh17s statement. Maybe less automation would be better here.

closed 2018-02-09 by thinkyhead