MarlinFirmware/Marlin #7406: Z_PROBE_END_SCRIPT executed after other GCode.

khcassidy, 2017-08-02:
Hello,
Running 1.1.3 RTM
I have a s small problem which has started happening: Z_PROBE_END_SCRIPT executed after other GCode. Gcode is executed out of order.

**gcode generated using Slic3r for starting a print**

_M107
M190 S65 
M109 T0 S205 
G28 ; home all axes
**G29**
;M82        ;set extruder to absolute mode
G1 Z0.2 X-70 Y-70 F10000
G92 E0
G1 X40 E20 F400             ; Extrude a fat line of filament to prime and clean the nozzle.
G92 E0                      ; Zero the extruder.
G1 E-2 F500                 ; Retract 1 mm.
G1 X80 F9000                ; Sweep right to cut any strings.
G92 E0_

**Marlin Z_PROBE_END_SCRIPT**
_#define Z_PROBE_END_SCRIPT "G90\nG1 X-88 Y65 Z30 F8000\nG1 X-88 Y65 Z18 F1500\nG1 X-80 Y55 Z45 F8000"_

All goes well till the bed leveling is completed (G29) at which point normally the Z_PROBE_END_SCRIPT is run after which I push some filament through the extruder at the front of the bed.

What has started happening for an unknown reason is that the G29 runs, it then it skips the Z_PROBE_END_SCRIPT, runs through the extruder priming gcode, AFTER which it decides to run the Z_PROBE_END_SCRIPT.

Any ideas why my gcode is running out of order?

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead

ChrisPVille, 2019-10-25:
I've had a similar problem for a few Marlin versions (Still is an issue for me on 1.1.9).  Both on the Printrbot simple metal original firmware (1.1.0ish) and the updated firmware (1.1.9), it first seemed like the results of the Z probe sometimes wouldn't apply to commands already in the queue, but I believe it was that the Z_PROBE_END_SCRIPT was queuing itself after the commands immediately following the G29.

They had defined it as `Z_PROBE_END_SCRIPT "G1 Z10 F7000\nG1 X0 Y0 F7000"`

As a test, I rebuilt the firmware without Z_PROBE_END_SCRIPT defined, and put the G1 moves in the print GCode immediately following the G29 (where they should go in either case).  The move behavior was correct only when Z_PROBE_END_SCRIPT was not used.

Again, if I had to take a guess at the cause, it would be that the Z_PROBE_END_SCRIPT is put on the command queue _after_ already issued commands:

Print GCode
Print GCode
Print GCode
G29 `  ;Z-Probe begins`
Print GCode  `  ;Queued command`
Print GCode  `  ;Queued command`
Print GCode  `  ;Queued command`
Z_PROBE_END_SCRIPT `  ;Put in queue at wrong spot`

thinkyhead, 2019-10-25:
We did not have facilities in Marlin 1.1.9 to run G-codes "in-place." So the only solution is to move to Marlin 2.0.