MarlinFirmware/Marlin #7419: Re-Arm and RRD Full Graphic Display

Grogyan, 2017-08-03:
Finally got my act together and got another RAMPS 1.4 from RepRap Discount
This board is used in conjunction with the Re-ARM board.

The RAMPS 1.4, adapter board and display are all white, not red.

On power up the display is blank and the buzzer continuously beeps.

Turns out that to be compatible with Re-ARM the plastic housings for the header pins, on the adapter board and Display need to be turned a complete 180 degrees.
Easing off the housings is not too hard, and requires a little bit of patience to prise off.

Cable orientation should be followed Panucatt's Re-ARM page
[Re-ARM to White RRD Full Graphic Display](http://panucattdevices.freshdesk.com/support/solutions/articles/1000243195-lcd-display-installation)

Bob-the-Kuhn, 2017-08-03:
Your LCD adapter came with the keying slots facing away from the controller??!!!

Most unusual.  Hope it's not a trend.

I saw the 180 degree rotation on the MKS Mini LCD display and on the LCD adapter for the RAMPS-FD.

Thank goodness that the RRDFG can stand being plugged in backwards.

Tannoo, 2017-08-03:
I was questioning that myself. One of the adapters I got had them backwards also. It my have fried the glcd. Although, I had not tested the lcd before plugging it in that incorrect adapter. (I didn't pay attention to it)

bobc, 2017-08-03:
I've seen a lot of the Chinese made stuff with connectors reversed, but
they supply reversed cables as well so all works out :) There is often no
pin 1 marked on the boards either, and two identical connectors, neither
marked... So much fun :) The lack of orientation and marking extends to the
schematics, so they don't help much either.
I'm not sure if they are deliberately obfuscating, or are just not bothered
about following minimal standards, I suspect the latter.

Grogyan, 2017-08-03:
I thought that too, that just turning around the cables will fix it.  that is until I remembered that the pin 10 on the adapter board side needs to be tied to a 5V rail, and reversing the cable is not a workable solution.

I'll put it on the wiki at some stage, about the work around of prising the plastic housings off and turning them around the other way.

fiveangle, 2017-08-09:
> I've seen a lot of the Chinese made stuff with connectors reversed

Yes, MKS LCDMINI12864 (minipanel) is like this too.

> I thought that too, that just turning around the cables will fix it. that is until I remembered that the pin 10 on the adapter board side needs to be tied to a 5V rail, and reversing the cable is [...]

How is shaving the key and plugging in 180 any different than prying the keyway seats off ?  The cable that comes with the Re-ARM has the 5v pin broken out on the board side so as long as you shave the key on the other end and flip, it's identical, no ? (I'm getting ready to plug mine in, and have said jacked-connector display so I want to understand what you're saying in case I'm missing something :)

Bob-the-Kuhn, 2017-08-09:
Shaving the key & reversing the cable works.

psavva, 2018-01-21:
@Grogyan, do you still have the issue with the buzzer?
I opened up this issue: https://github.com/MarlinFirmware/Marlin/issues/9279

I'm facing the same w.r.t the Buzzer sounding.
I shaved the key and reversed the connector.

The LCD and SD works perfectly otherwise...

Grogyan, 2018-01-21:
I do not have the issue of the buzzer buzzing since reversing the connector housing.

My experiments are currently halted, waiting for SPI fix for the MAX31855 #8010

gavatron3000, 2018-01-25:
Thanks for this, i had the same problem. working now

thinkyhead, 2018-02-11:
Is this a problem with your specific RAMPS shield, or does this affect all Re-Arm boards?

gavatron3000, 2018-02-11:
My plugs were reversed compared to the ramps pictured in panucatt LCD cable  procedure

closed 2018-02-19 by thinkyhead