MarlinFirmware/Marlin #7423: [FR] Adding Arabic language characters in Marlin Software

bathnsahin, 2017-08-03:
Hi, 

I try to add Arabic language my LCD screen. But i couldn't. I looked many issues, i tried some methods. But i can't see clearly Arabic characters on LCD. I found Arabic's ISO number but i couldn't find Mapper for it. Please help me :(

petrzjunior, 2017-08-04:
Some letters (Latin, Japanese and Russian) are supported on the "text" LCD, but they are hardcoded and can't be changed. On the Graphical LCD you can draw basically anything, but you need the correct font. There are already some mappers for Chinese, Polish and Czech characters, but now Marlin is awaiting total reorganization and hopefully also automatic UTF-8 mapper, so no new manual mappers are being added.

bathnsahin, 2017-08-04:
Thank you. I found some arabic fonts but they don't work. I hope Marlin will do this quickly. @petrzjunior

thinkyhead, 2018-02-09:
> I hope Marlin will do this quickly.

Alas, none of our more active contributors speaks Arabic. I blame Abu Hamid al-Ghazali.

boelle, 2019-02-19:
@bathnsahin you need to find some people that speak Arabic before this can happen

boelle, 2019-06-22:
maybe add feature request label

IbrahemAlbanawi, 2022-12-13:
There was a 3D printer with an Arabic interface unfortunately the company went out of business and they don't provide the firmware file I don't own one but there's someone who does and I'm no expert on how to extract the firmware and see how they did it so it's definitely possible