MarlinFirmware/Marlin #7435: latest bugfix-1.1.x still puts UBL mesh outside of the bed

040medien, 2017-08-05:
I was very excited to learn about the new bed size parameters. Since my homing coordinates are off the bed and my bed is not symmetrically located between the physical travel limits, I thought that this way finally Marlin will be able to overlay the grid perfectly over the actual bed.

However, it looks like the UBL_MESH_INSET start from [X_MIN_POS, Y_MIN_POS], not from [0, 0]. I don't think that makes sense, since the G26 test pattern is then also printed off the physical bed dimensions.

My settings:

```// The size of the print bed
#define X_BED_SIZE 207
#define Y_BED_SIZE 312

// Travel limits after homing (units are in mm)
#define X_MIN_POS -20
#define Y_MIN_POS -2
#define Z_MIN_POS 0
#define X_MAX_POS X_BED_SIZE
#define Y_MAX_POS Y_BED_SIZE
#define Z_MAX_POS 212

 #define UBL_MESH_INSET 10          // Mesh inset margin on print area
```

Roxy-3D, 2017-08-05:
Yes.   I understand your point.   It will take some time to get UBL synchronized to the BED_SIZE parameters.

040medien, 2017-08-05:
Looks like this can be resolved by changing these values:
```#define UBL_MESH_MIN_X (UBL_MESH_INSET)
#define UBL_MESH_MIN_Y (UBL_MESH_INSET)```

I think these should be the defaults.

040medien, 2017-08-10:
@Roxy-3D do you think my suggestion above is reasonable? At least for cartesian printers? I was able to confirm that it works as expected for me.

Roxy-3D, 2017-08-10:
This is an interesting idea.   The problem is I don't fully understand where all the impact is yet for the changes caused by switching to X_BED_SIZE & Y_BED_SIZE.   That is the reason for no response.    But one thought I have is in the 'normal' ?_BED_SIZE setup, the origin is assumed to be at (0,0).   In that case

```cpp
  #define UBL_MESH_MIN_X (X_MIN_POS + UBL_MESH_INSET)
```
the current setup ends up being exactly what you propose because X_MIN_POS is 0.0.   

Are you using the X_BED_SIZE & Y_BED_SIZE values to define your bed?    (Right now we are in a transition period so if not...  it would be helpful for you to shift to the X_BED_SIZE & Y_BED_SIZE mindset. )

I'm very open to what ever ideas you have to 'fix' this!

040medien, 2017-08-10:
Thanks for the response and keeping an open mind. Yes, I'm using X_BED_SIZE & Y_BED_SIZE as you can see in my original post above.

I actually disagree with your premise. X_MIN_POS + UBL_MESH_INSET != 0

Here's my interpretation of how this works or should work for a cartesian printer:
- * The physical bed is delimited by [0, 0] and [X_BED_SIZE, Y_BED_SIZE]
- * The inset starts from the these above dimensions inward and is independent from the travel limits. The reasons we have it is to have enough material underneath to probe, because the probe itself has a diameter and sometimes since we want to stay clear of the bed screws or clamps, etc.
- * X_MIN_POS != X_BED_SIZE - X_MAX_POS, i.e. the bed center and the center between the travel limits are not the same

With that in mind, the mesh should only span the bed (minus the inset). The travel limits are only considered together with the probe offsets to understand where the probe can go, not where the mesh should be.

To illustrate, I created this image which shows a 3x4 grid in a setup similar to mine: https://ibb.co/jCYwNF

Roxy-3D, 2017-08-10:
You make a good argument...    It is going to take days to work through all the issues.   It might make sense to make the suggested changes and see if it goes smoothly for everybody.  Can you generate a Pull Request?

closed 2018-02-09 by thinkyhead