MarlinFirmware/Marlin #7436: Z Babysteps Not Applied to Delta During Print

tcm0116, 2017-08-05:
Using 1.1.x-bugfix as of a few days ago, I'm not seeing the Z axis babysteps being applied to my delta during the print while adjusting the Z offset. If I run `G29` and then adjust the Z offset, I can see the nozzle moving up and down. However, once I start a print, there is no effect when I adjust the Z offset.

For reference, I'm using bilinear leveling and a BLTouch, however I don't think those are contributing factors since the babysteps are applied when there is no other motion while the leveling is active.

ImplementOfWar, 2017-08-11:
Is your mesh active?  For example every time you home with G28 it will disable the mesh.  You can M420 S1 to turn "bed leveling" back on.  You should uncomment #define BABYSTEP_ZPROBE_OFFSET since you are using auto leveling and a z probe.

tcm0116, 2017-08-11:
My start GCode always calls G29 immediately after calling G28, so the mesh should be active. It's hard to tell for sure since I came just watch the Z axis like on a Cartesian machine. 

BABYSTEP_ZPROBE_OFFSET  is enabled. If I run G28 followed by G29, I can adjust the Z Offset and the nozzle moves up or down accordingly. However, there appears to be no change to the Z height if I adjust the Z Offset during a print. 

I looked at the babystepping code for Delta machines, and I didn't see anything obvious as you why it would work when there are no other movements occurring, but not work during printing.

ImplementOfWar, 2017-08-11:
G29 doesn't turn the mesh in after homing,  it just tells you what the mesh is and lets you set mesh points.  You need to run m420 after g28

tcm0116, 2017-08-11:
I think that's true for UBL, but I'm using bilinear which should be active
immediately upon completion of G29. In any case, you're missing my point. I
can adjust the Z height via babystepping when there are no other queued
movements, but not when during a print.

ImplementOfWar, 2017-08-11:
Incorrect.  It is true for BILINEAR bed leveling.  You must have a really level bed and very small mesh corrections to not notice your mesh is not being applied in the first place.  This can be solved in 5 seconds if you just at least try solutions people are offering you,  You most likely have a startup script for each print that homes an axis before it starts,  which disables the mesh.  In your slicer script for when a print starts,  put in M420 S1 after any homing commands if there are any and then test Z babystepping.  Good luck with your issue.

tcm0116, 2017-08-12:
I appreciate you help, but we appear to be having some communication issues
here as you keep telling me that G28 turns off the bed leveling, and I keep
telling you that my start script for every print calls G28 immediately by
G29, and then the print starts. Since I'm leveling *before* every print,
the mesh should be active as soon as the G29 completes.

I can prove this because I can turn on my printer, execute G28 and then
G28, and then subsequent adjustments to the Z Offset result in the nozzle
moving up or down accordingly. As the author of BABYSTEP_ZPROBE_OFFSET, I
know for a fact that the babysteps are only applied when ABL is active.

I've now switched to linear leveling, and BABYSTEP_ZPROBE_OFFSET is working
as expected. My guess is that there's some interaction between babystepping
and bilinear leveling that is causing them to cancel each other out. In
other words, the babysteps are being applied, but bilinear leveling is
assisting the height of the nozzle in response to the Z offset changing,
resulting in a net 0 change in the nozzle height. This is just a theory at
this time as I haven't had an opportunity to look into it. I seem to
remember an issue recently where someone was saying that the nozzle was
going the wrong direction when adjusting the Z Offset, so perhaps whatever
change was made to fix that broke BABYSTEP_ZPROBE_OFFSET for bilinear
leveling.

thinkyhead, 2018-02-09:
Fixed for 1.1.9.

closed 2018-02-09 by thinkyhead