MarlinFirmware/Marlin #7438: [FR] Add support for RealTimeClock and uptime reporting

bipsendk, 2017-08-05:
Sometime it might become handy to be able to retrieve uptime information from the printer - and if no other timers are handling it, and might be achieved if a RTC (RealTimeClock) chip is connected to the controller board.
Some board have I2C capabilities (RAMPS among others) - and adding for example a module with a DS1307 RTC (and often also a AT24C32  EEPROM), will make it possible to set the time on the printer using the M905 command.
Uptime (since last reset/power-on) could be read out via the menu - or maybe in the printer firmware message (M115).
Whenever the printer is reset/rebooted, the currect timestamp can be read from the RTC and stored in EEPROM. When uptime info is needed, it is a question of reading the RTC and subtract the stored timestamp - thereby having the uptime available.

boelle, 2019-02-19:
i cant see the benefit of uptime or timestamps

i dont even look at serial after rpi3 and octoprint came to see the day of light

7eggert, 2019-04-06:
Uptimes / Timestamps have as much benefit as RGB LED. But I just encountered a reason to use the RTC / external EEPROM: One of the users of my favorite forum seems to ave a b0rken EEPROM.