MarlinFirmware/Marlin #7439: msg too cold for M600 missing

laurent1991, 2017-08-05:
hello guys,

I think there is a problem (not sure) and sorry for my bad english.
- Description: M600 msg too cold for M600 missing
in the language files, 'M600' is missing (MSG_TOO_COLD_FOR instead of MSG_TOO_COLd_FOR_M600)

Have a nice day.

Roxy-3D, 2017-08-05:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/language.h#L196

Yes.  I see what you mean.   We may need to change this:
```cpp
  inline void gcode_M600() {

    if (!DEBUGGING(DRYRUN) && thermalManager.tooColdToExtrude(active_extruder)) {
      SERIAL_ERROR_START;
      SERIAL_ERRORLNPGM(MSG_TOO_COLD_FOR_M600);
      return;
    }
 ```

tcm0116, 2017-08-06:
This is fixed in #7107

closed 2018-02-09 by thinkyhead