MarlinFirmware/Marlin #7440: extra space between Z-layers leaves print weak

nealpalmer, 2017-08-05:
### Bug Report
- Description: extra space between layers
- Expected behaviour: all layers should have the same thickness
- Actual behaviour: about 1 in 10 prints will have a group of about 3 layers in a row that appear to be thicker than normal.  I'm not sure if this is because of the z-axis moving too far, or no filament extruded for the layer, or the previous layer got smashed down to no height, and the printer then added that extra height over the next few layers.  This didn't happen when I was using sprinter firmware, but I switched to Marlin because of the bed leveling.
- Steps to reproduce:
  - Print a bunch of stuff and sometimes a group of layers will be too thick (visible on the side of the print because it is slightly inset.  The place in the print isn't consistent.  The added height isn't consistent (sometimes about 0.25mm, sometimes about 1mm).

Attach a ZIP of `Configuration.h` and `Configuration_adv.h` by dropping here.

------------------------------------------------------------
### Bug Report Tips
- When troubleshooting, use `M502` followed by `M500` to reset EEPROM to defaults.
- Use `DEBUG_LEVELING_FEATURE` with `M111 S247` for detailed logging of homing/leveling.
- Format text with: **bold**, _italic_, `code`.
- Format C++ with three backticks, plus "cpp":
```cpp
void my_function(bool do_it) {
  // Hold this spot
}
```

nealpalmer, 2017-08-12:
More information.  I just watched/heard it do a major Z-axis change in the middle of printing infill on a layer.  Any idea why this happens?

I also got a print that finished, and it is about 1mm taller than other prints from the same gcode file, and it has about a 1mm thicker layer in the middle.

Roxy-3D, 2017-08-12:
What bed leveling system are you using?   

There have been similar reports.   Does your mesh have a lot of numbers that are close to -1.00 mm in value?

nealpalmer, 2017-08-14:
Bilinear Leveling Grid:
      0      1      2      3
 0 -0.102 +0.050 +0.254 -0.054
 1 -0.159 -0.113 +0.010 -0.133
 2 -0.314 -0.074 +0.065 -0.228
 3 -0.566 -0.254 -0.100 -0.392

Definitely not a stepper driver overheating.  I heard the z-motion in the middle of infill (not sure about my occasional Y-shifts, because I'd assumed they were missed steps).

Looks like the same problem as #7009.

I just posted this to 7009:
I'm having the same problem on 1.1.1. But I usually see the problem on the Z-axis showing extra thick layers that aren't bonded well to the layer below and are like spaghetti (about 10% of prints do this once or twice). I heard it happen once: it was in the middle of a layer doing infill, and the x/y stopped moving, and Z did a large move (about 1mm) which was very audible, then the print separated after that. I think I have also seen the Y-axis shift, but much more seldom than the z-axis. My problem is definitely not the stepper driver overheating.

nealpalmer, 2017-08-17:
Any idea how to add debugging information in the right part of the code to catch the issue?

thinkyhead, 2018-02-09:
Presuming solved.

closed 2018-02-09 by thinkyhead