MarlinFirmware/Marlin #7441: Not Working: Need different steps for dual extruder

ahsnuet09, 2017-08-05:
I am actually facing a problem that I need different steps_per_unit for each extruder but unfortunately, Marlin only considers the first value of steps_per_unit given for extruders and operate both extruders according to that value.

it operates both extruders according to first given value **{99}** and don't consider the second value for second extruder **{1320}**
**#define DEFAULT_AXIS_STEPS_PER_UNIT {87.5, 87.67, 2520.49, 99, 1320} // X,Y,Z,E0...**

I am using // RAMPS 1.3 = 34 (Power outputs: Extruder0, Extruder1, Bed)

Looking forward to your response......

Kaibob2, 2017-08-06:
Do you have [Distinct_E_factors](https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/Configuration.h#L505) uncommented?

ahsnuet09, 2017-08-08:
Thank You @Kaibob2
I overlooked that before,,,,,it works now

Kaibob2, 2017-08-08:
Great. Please close here then.

closed 2017-08-09 by ahsnuet09