MarlinFirmware/Marlin #7443: Marlin turns on pullup on TEMP_0_PIN (ANET_10)

emente, 2017-08-06:
### Bug Report
ANet 3D 1.0 Board Temperature reading issue, Marlin 1.1.x origin head, 05.07.2017, d3bc877007ceaeaeed11898998ee83b54f83de46

- Description: It seems Marlin turns on pullup on TEMP_0_PIN (pin A7 = PA7 of 1284P), corrupting temperature readout of extruder 0
- Expected behaviour:  extruder temp 0 reads out correctly
- Actual behaviour: TEMP_0_PIN reads 0°C (due to pullup?)
- Steps to reproduce:
  - flash marlin (ANET_10 board define, arduino ide 1.6.9, Board Mighty Mini 1284P 16 Mhz Optiboot)
  - Turn on, keep reset pressed, Voltage at Pin A7 is about 4.72V. Voltage is caused by 4.7K pullup + ATC Semitec 104GT-2 (reading 98K) and looks correct as TEMP_BED_PIN reads 4.73V.
  - Stop Resetting. When Marlin initializes Voltage at Pin A7 jumps to 4.99V, causing read temperatue to be 0°C
  - Everything works (including zmin autoleveling and heatbed), but if i raise the temperature on the extruder the temperature protection is triggered because of 0°C reading

[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1203029/Marlin.zip)

I have no idea where to search or how to debug this. Any ideas?

emente, 2017-09-13:
compiling without sdcard support resolves the problem with the pullup.

doubly used pins for anet 3d?

Phr3d13, 2017-10-09:
You may have a poorly soldered board, did you see any bad joints while you were examining the board?

Phr3d13, 2017-10-12:
just noticed something... "Board Mighty Mini..." is that the board you picked in configuration.h, or in arduino?

closed 2017-11-15 by thinkyhead