MarlinFirmware/Marlin #7445: UBL error z_correction

calvinwhitsell, 2017-08-06:
### Bug Report
- Description: Recieve the following example errors
11:13:32.689 : yl_i out of bounds in z_correction_for_y_on_vertical_mesh_line(ly0=299.00, xi=1, y1_i=9)
11:18:27.261 : yi out of bounds in z_correction_for_x_on_horizontal_mesh_line(lx0=375.00,x1_i=9,yi=1)

Additionally have a nozzle height issues similar to #7444

- Expected behaviour: The UBL feature prints on the bed.

- Actual behaviour: The UBL seems to "float" 1mm over the bed.

- Steps to reproduce:
  - Follow the steps in the wiki http://marlinfw.org/docs/features/unified_bed_leveling.html
  - When I get to the G26 step "g26 b60 h215 f1.75 l0.2 n0.4" I end up with the nozzle running around the bed about 1 mm so the first layer has no adhesion. 

Printer, Rigidbot Big with the Rigidboard v2. E3d dual hot end.
[config.zip](https://github.com/MarlinFirmware/Marlin/files/1203240/config.zip)

G29 W
`11:29:08.845 : N48 G29 W*89
11:29:08.868 : Unified Bed Leveling System v1.01 active.
11:29:08.917 : Mesh 0 Loaded.
11:29:08.964 : UBL object count: 1
11:29:08.986 : planner.z_fade_height : 10.0000
11:29:08.986 : zprobe_zoffset: -0.0100000
11:29:08.986 : UBL_MESH_MIN_X  (0 + 1)=1
11:29:08.986 : UBL_MESH_MIN_Y  (0 + 1)=1
11:29:09.002 : UBL_MESH_MAX_X  (376 - (1))=375
11:29:09.018 : UBL_MESH_MAX_Y  (300 - (1))=299
11:29:09.034 : GRID_MAX_POINTS_X  10
11:29:09.034 : GRID_MAX_POINTS_Y  10
11:29:09.066 : MESH_X_DIST  41.56
11:29:09.066 : MESH_Y_DIST  33.11
11:29:09.354 : X-Axis Mesh Points at: 1.000  42.556  84.111  125.667  167.222  208.778  250.333  291.889  333.444  375.000
11:29:09.610 : Y-Axis Mesh Points at: 1.000  34.111  67.222  100.333  133.444  166.556  199.667  232.778  265.889  299.000
11:29:09.611 : Kill pin on :32  state:1
11:29:09.675 : ubl_state_at_invocation :0
11:29:09.675 : ubl_state_recursion_chk :0
11:29:09.723 : Meshes go from 0x0280 to 0x0F7F
11:29:09.771 : sizeof(ubl) :  1
11:29:09.771 : z_value[][] size: 400
11:29:09.803 : EEPROM free for UBL: 0x0CFF
11:29:09.850 : EEPROM can hold 8 meshes.
11:29:09.883 : Unified Bed Leveling sanity checks passed.
`

G29 T
`11:22:11.883 : Bed Topography Report for CSV:
11:22:12.124 : -0.470,-0.510,-0.074,-0.002,0.067,0.069,0.010,-0.089,-0.184,-0.289
11:22:12.316 : -0.520,-0.570,-0.041,0.047,0.118,0.127,0.086,0.001,-0.089,-0.179
11:22:12.467 : ,0.083,0.158,0.202,0.167,0.086,0.020,-0.051
11:22:12.628 : -0.520,-0.550,0.002,0.108,0.201,0.237,0.218,0.178,0.112,0.068
11:22:12.788 : -0.600,-0.570,0.026,0.142,0.236,0.299,0.271,0.240,0.192,0.148
11:22:13.012 : -0.640,-0.600,0.053,0.172,0.269,0.319,0.313,0.296,0.246,0.202
11:22:13.156 : -0.620,-0.620,0.066,0.189,0.291,0.348,0.357,0.331,0.282,0.241
11:22:13.316 : -0.600,-0.650,0.097,0.212,0.326,0.391,0.401,0.368,0.320,0.276
11:22:13.473 : ,-0.650,-0.680,-0.690,-0.620,-0.570,-0.480,-0.440
11:22:13.634 : -0.640,-0.680,-0.650,-0.690,-0.690,-0.680,-0.620,-0.580,-0.500,-0.460
`

Please let me know if you need any further information.
Thank you for any assistance.
Regards,
Calvin

Roxy-3D, 2017-08-06:
Those error messages are because you tried to print off of the mesh.  The mesh is not big enough to provide correction information for the size of the print that you are doing.

calvinwhitsell, 2017-08-06:
Hello Roxy,

Any reason this is printing off the mesh, I am not aware of what controls the size of the G26 print..

Regards,
Calvin

Roxy-3D, 2017-08-06:
In you slicer...  you can scale the size of the print.   If you scale the object to 1/2 its size, it probably won't go off the mesh.   Those 'off the mesh' messages are harmless.   The only bad thing is you won't be getting any Z-Height correction there so you may not get good adhesion.

Also...   Most of the slicers document the object and the slicer settings used to generate the GCode inside the GCode file.   And usually both the Slicer and control program will tell you how big the print is going to be.   For example...  pulling a random .GCode file into PronterFace tells me this:

Loading file: C:\Downloads\3D-Printers\Files-for-i3\3 Parts\Layer_Print\layer_with_rings_and_support.gcode
Loaded C:\Downloads\3D-Printers\Files-for-i3\3 Parts\Layer_Print\layer_with_rings_and_support.gcode, 355279 lines
34515.69mm of filament used in this print
The print goes:
- from 90.55 mm to 284.44 mm in X and is 193.89 mm wide
- from 102.13 mm to 272.87 mm in Y and is 170.74 mm deep
- from 0.00 mm to 67.60 mm in Z and is 67.60 mm high
Estimated duration: 346 layers, 4:44:54

If any of those bed dimensions are outside of what you have declared the position of the UBL Mesh, you are going to go off the mesh.

calvinwhitsell, 2017-08-09:
Hello @Roxy-3D ,

Thank you for the details. I was using repeater-host to send a g26 for printing so I am not sure how to prevent those errors since I didn't slice anything or load a file. Would you have any suggestions on this or would that be something I would need to research with repeater?

Regards,
Calvin

Roxy-3D, 2017-08-09:
> I was using repeater-host to send a g26 for printing so I am not sure how to prevent those errors since I didn't slice anything or load a file. 

This is a little bit controversial.   It is perfectly possible to generate an .STL file with what ever 'Mesh Validation Pattern' is appropriate, run that through a Slic3r and print it.  But it is a lot easier on the user to just send a bare minimum G26 P C O2.0 and see what the printer can do....    And if it doesn't do the 'right' thing, you just edit the mesh  based on what you see.

G26 doesn't need a file.  It knows enough to use your correction mesh and print a pattern so you can directly edit the mesh based on the results.

In PronterFace, you would just type G26 P C O2.0  (and maybe a few more parameters to adjust a couple of things) and see what happens.

FHeilmann, 2017-08-09:
What I did before G26 was implemented:

In your slicer, import a cube, scale it so it's dimensions are `<bed size x>*<bed size y>*<layer height>`

set top and bottom layers to 0 and give it an infill style and percentage. I use ~20% infill with honeycomb.

This will give you a somewhat usable g-code to check for issues with your bed leveling, the only disadvantage compared to G26 is that you won't have the "crosshairs" over the probe points.

Here's a picture of what one of my printouts looked like:
[image]

lrpirlet, 2017-08-09:
Hi @calvinwhitsell 
I am printing from Repetier... 
I did issue G26 and got a very nice print without any error at all... have a look at the 2 last entries of #7021

Hint: make sure to use S0.04  instead of N0.04 if using one of the latest rev of Marlin... doc may NOT be in tune with current version...

lrpirlet, 2017-08-09:
Ooops, sent that too fast...

Second hint: make sure Repetier is NOT setup to touch the Z value ... there is a Repetier parameter that will change Z height when printing... There is NO INFLUENCE of this parameter over the G26... and that is the beauty of that G26 code :-)

Enjoy.

Roxy-3D, 2017-08-09:
> This will give you a somewhat usable g-code to check for issues with your bed leveling, the only disadvantage compared to G26 is that you won't have the "crosshairs" over the probe points.

Yes, it is true it doesn't give you the 'crosshairs' to pin point the mesh locations.   But it almost wouldn't matter.    You can move the nozzle to the center of an area that needs refinement and issue the G29 P4 R to edit those mesh points.   The nozzle will move to each mesh point in the area and the user can edit it.

This is a very viable way to get the G26 functionality without building the G26 code into the firmware image.

closed 2017-11-03 by thinkyhead